smartdevicelink/sdl_core #631: What is CI/TeamCity??

JackLivio, 2016-06-21:
I have a pr that is not passing the CI/TeamCity Winx64 build. If i click on details it takes me to a private IP address I can't access. 

How are developers supposed to fix their build failures if the CI server is private?

AByzhynar, 2016-06-22:
@JackLivio Now you can go through `Details` links in pull request. Old links were leading to private IP, all new links are leading to public IP, which is accessible from everywhere. 
P.S. If you need to re-run you PR in some configuration( Windows or QT) you have to register TeamCity account(via the same link). Just Add your real name and email (to avoid to be deleted as unknown user).
You can also login as guest but in this case you will be able to see build logs and nothing more.

Best regards,

JackLivio, 2016-06-23:
@AByzhynar Thank you

closed 2016-06-23 by JackLivio

justinjdickow, 2016-06-23:
@CMaylone fyi