smartdevicelink/sdl_core #636: Policy update is been triggered Continiously if we enable  flag ENABLE_EXTENDED_POLICY

madhuy, 2016-06-22:
1) Enabled ENABLE_EXTENDED_POLICY in SDL core.
2) Policy update been triggered continuously 

Please look for "BasicCommunication.PolicyUpdate" in logs. 

[SmartDeviceLinkCore.txt](https://github.com/smartdevicelink/sdl_core/files/328778/SmartDeviceLinkCore.txt)

JackLivio, 2016-06-23:
@madhuy looking into this

JackLivio, 2016-06-23:
@madhuy Do you see this behavior every time? 

I am unable to replicate this behavior at the moment. Using recent versions of SPT, Core, and the web hmi I only see the initial policy update.

JackLivio, 2016-06-23:
I also do not see BasicCommunication.OnSystemRequest in your logs. I think you might have something hooked up wrong in your HMI?

madhuy, 2016-06-23:
@JackLivio 
with ENABLE_EXTENDED_POLICY it calls the  "BasicCommunication.PolicyUpdate" NOT "BasicCommunication.OnSystemRequest" for policy update

I think I know the reason . We need to call 
 policy::PolicyHandler::instance()->OnUpdateRequestSentToMobile(); to reset update_required_ value.

in [SendPolicyUpdate](https://github.com/smartdevicelink/sdl_core/blob/master/src/components/application_manager/src/message_helper.cc#L1580)()

With this fix it is working I will create PULL request.

madhuy, 2016-06-23:
@JackLivio  Please review change https://github.com/smartdevicelink/sdl_core/pull/644/files

AnastasiyaBritanova, 2016-07-08:
Hi Guys!

Please - it seems you'll break the expected flow of events with these fixes :)

Correct flow for ENABLE_EXTENDED_POLICY: **ON**

1. policies update is triggered: SDL -> HMI: PolicyUpdate (with Snapshot of current policies)
2. HMI **must request** the policices urls via GetURLs RPC 
3. Afterwards, **HMI must generate OnSystemRequest** through SDL to mobile app.

You should not fix SDL to generate OnSystemRequest - in this case, it's HMI's responsibility.

AnastasiyaBritanova, 2016-07-08:
Please see the attached diagram for details.
[image]

JackLivio, 2016-07-12:
@madhuy see proper flow above

JackLivio, 2016-07-12:
@madhuy After the HMI sends an OnSystemRequest to Core, [the application manager will call](https://github.com/smartdevicelink/sdl_core/blob/master/src/components/application_manager/src/application_manager_impl.cc#L1269) 
```
OnUpdateRequestSentToMobile();
``` 
when processing the OnSystemRequest message from the HMI before passing along the message to the mobile side.

madhuy, 2016-07-12:
@JackLivio  I understand the Flow.  in our HMI, HMI does not create BC.OnSystemRequest to get Policy data from server.  we get Policy data with different mechanism(secure channel), store into the file and directly call the SDL.OnRecievedPolicyUpdate() .  

my question , where the right place to set s 'update_required_' to 'false' in this case.

justinjdickow, 2016-07-13:
@madhuy Will test with using `set_update_required(false);` in `UpdateStatusManager::OnValidUpdateReceived()`

joeygrover, 2016-08-08:
Current PR open and being discussed on #644

jacobkeeler, 2016-10-06:
New PR: #865

theresalech, 2016-10-19:
Pull request #865 was merged to fix this issue.

closed 2016-10-19 by theresalech