smartdevicelink/sdl_core #647: RegisterAppInterface returns WARNING message if app has AppHMIType

madhuy, 2016-06-27:
1. Added HMItype in app policies
2. send AppHMIType from mobile app 

RegisterAppInterface responses WARNING with message
""info" : "Following AppHMITypes are not present in policy table:5,""

justinjdickow, 2016-07-07:
Can you send a snapshot of the policy table you're using and the AppHMIType that was defined by the mobile side?

AnastasiyaBritanova, 2016-07-08:
From requirements point of view, SDL returns RAI (WARNINGS) in case appHMITypes from RAI **partially** match appHMITypes from app's policies.

To receive RAI (SUCCESS) you need to make sure both RAI and app's policies have absolutely the same appHMITypes.

_Note:_ appHMITypes must be from "AppHMIType" enum of mobile_API only.

madhuy, 2016-07-08:
here is  the App policy I am using.
{
        "keep_context": false,
        "steal_focus": false,
        "priority": "NONE",
        "default_hmi": "NONE",
        "AppHMIType": [ "INFORMATION" ],
        "groups": [
          "Base",
          "Location",
          "Notifications",
          "DrivingCharacteristics",
          "VehicleInfo",
          "OnKeyboardInputOnlyGroup",
          "OnTouchEventOnlyGroup",
          "SendLocationOnly",
          "OnButtonPressOnly"
        ]
      }

AnastasiyaBritanova, 2016-07-11:
If your app sends 
<RAI ("AppHMIType": [ "INFORMATION" ])> 
and you are still getting 
<WARNING> with message <""info" : "Following AppHMITypes are not present in policy table:5,""> 
- it's a defect and needs to be investigated in the SDL code.

justinjdickow, 2016-07-12:
@madhuy Can you confirm so we can label this a bug and move forward with that investigation?

madhuy, 2016-07-13:
Yes , We are getting ""info" : "Following AppHMITypes are not present in policy table:5,"
and it is bug.

jacobkeeler, 2016-07-14:
@madhuy I am having some difficulty reproducing the issue using the steps you gave, could you please post the RegisterAppInterface request that core receives from the mobile side?

madhuy, 2016-07-14:
Message from registerappinterface

Message to convert: protocol 2; json {"languageDesired":"EN-US","deviceInfo":{"osVersion":"9.2.1","os":"iPhone OS","carrier":"Verizon","hardware":"iPhone"},"appName":"CTS Policy Table App","appID":"000000005","appHMIType":["INFORMATION"],"hmiDisplayLanguageDesired":"EN-US","vrSynonyms":["C T S Policy Table"],"ngnMediaScreenAppName":"CTS Policy Table App","syncMsgVersion":{"majorVersion":1,"minorVersion":0},"ttsName":[{"type":"TEXT","text":"CTS Policy Table App"}],"isMediaApplication":false}

attaching complete logs.
[SmartDeviceLinkCore-1.txt](https://github.com/smartdevicelink/sdl_core/files/364413/SmartDeviceLinkCore-1.txt)

AnastasiyaBritanova, 2016-07-15:
@madhuy just to make sure all of preconditions are correct - 
your app has the following representation in policies, right? (meaning: is placed in the correct section with the correct appID)
```
{
  "policy_table": {
...
    "app_policies": {
      "000000005": {
"keep_context": false,
"steal_focus": false,
"priority": "NONE",
"default_hmi": "NONE",
"AppHMIType": [ "INFORMATION" ],
"groups": [
"Base",
"Location",
"Notifications",
"DrivingCharacteristics",
"VehicleInfo",
"OnKeyboardInputOnlyGroup",
"OnTouchEventOnlyGroup",
"SendLocationOnly",
"OnButtonPressOnly"
]
}
```

jacobkeeler, 2016-07-15:
Found the problem. [This line](https://github.com/smartdevicelink/sdl_core/blob/bcf8405bbcb80823ac8bda564115695826ca99c4/src/components/application_manager/src/commands/mobile/register_app_interface_request.cc#L614) retrieves the list of appHMITypes from the received message, but as it turns out, the elements in this array are in enum form, while the program seems to assume they are in string form.
```cpp
smart_objects::SmartArray app_types = *(message[strings::msg_params][strings::app_hmi_type].asArray());
```

Meaning that in your case, it received an array like this: `[5]`. The array retrieved from the policy table was `["INFORMATION"]`, so when it compared the two arrays, they obviously did not match.

I'm not certain whether the types in the array were in enum form intentionally, but this issue is easily fixed by adding a conversion function from AppHMIType to string like this

```cpp
std::string AppHMITypeToString(mobile_apis::AppHMIType::eType type) {
  if (mobile_apis::AppHMIType::DEFAULT == type) {
    return "DEFAULT";
  } else if (mobile_apis::AppHMIType::COMMUNICATION == type) {
    return "COMMUNICATION";
  } else if (mobile_apis::AppHMIType::MEDIA == type) {
    return "MEDIA";
  } else if (mobile_apis::AppHMIType::MESSAGING == type) {
    return "MESSAGING";
  } else if (mobile_apis::AppHMIType::NAVIGATION == type) {
    return "NAVIGATION";
  } else if (mobile_apis::AppHMIType::INFORMATION == type) {
    return "INFORMATION";
  } else if (mobile_apis::AppHMIType::SOCIAL == type) {
    return "SOCIAL";
  } else if (mobile_apis::AppHMIType::BACKGROUND_PROCESS == type) {
    return "BACKGROUND_PROCESS";
  } else if (mobile_apis::AppHMIType::TESTING == type) {
    return "TESTING";
  } else if (mobile_apis::AppHMIType::SYSTEM == type) {
    return "SYSTEM";
  } else {
    return NULL;
  }
}
```

Which we can use to convert the enum when we compare the arrays, changing [this line](https://github.com/smartdevicelink/sdl_core/blob/bcf8405bbcb80823ac8bda564115695826ca99c4/src/components/application_manager/src/commands/mobile/register_app_interface_request.cc#L96)
```cpp
std::string app_type_str = value.asString();
```
to
```cpp
std::string app_type_str = AppHMITypeToString(((mobile_apis::AppHMIType::eType)value.asUInt()));
```

PR #702 (2016-08-04): Fix WARNINGS response when AppHMIType received
When comparing the lists of AppHMITypes from the policy table and RegisterAppInterface message, the values are mistakenly read from the message object as strings when they are actually in enum form, resulting in a warning even if both lists include the same AppHMIType. 

This PR changes this so that the enum values are converted to strings when comparing the two lists, so that they are compared correctly.

Fixes #647
closed 2016-08-04 by anosach-luxoft