smartdevicelink/sdl_core #665: USB Device is detected but apps are not listed

MayilrajC, 2016-07-02:
I have compiled the SDL core with Qt.
When i connect using USB, I was able see device name listed, but apps are not listed.

justinjdickow, 2016-07-13:
@MayilrajC What SDL applications are on the device which support USB?

MayilrajC, 2016-07-14:
I have Pandora and iHeartRadio installed in the phone.

MayilrajC, 2016-07-22:
Can anyone tell me list of iPhone apps, which supports smartdevicelink.

BrandonHe, 2016-08-18:
@MayilrajC You can try download app **Ford App Catalog** from App Store, in this app, all support SDL apps are listed, and it will keep update if new apps are added.

raspeeyan, 2016-09-07:
@justinjdickow 

I am also experiencing the same problem. I have created a sample application which SDL Core detects via Bluetooth, but cannot through USB. SDL can detect the Android device, however.

I have added the needed permissions for Android Accessory in the manifest file. What else do I have to set up?

Much thanks!

JackLivio, 2017-04-18:
I will reopen if people are still having this issue

closed 2017-04-18 by JackLivio

ashokvasarla, 2017-08-15:
USB Device is detected and connected but apps are not listed.
I have registered a test APP with new SdlProxyALM(this, "testapp", false, usbtransport);
I observed this issue is still exists and i see the same code MayilrajC commented on Jul 2, 2016.