smartdevicelink/sdl_core #674: Getting error during sdl_core make when installing with Qt

sztankatt, 2016-07-06:
When following this guide: https://github.com/smartdevicelink/sdl_core/wiki/SDL-on-Linux-with-QT, I get an error whe I try to make the project. 

The error output is the followig:
`/home/tst/Documents/Projects/smartdevicelink/sdl_core/src/components/application_manager/src/commands/hmi/dial_number_response.cc: In member function ‘virtual void application_manager::commands::hmi::DialNumberResponse::Run()’:
/home/tst/Documents/Projects/smartdevicelink/sdl_core/src/components/application_manager/src/commands/hmi/dial_number_response.cc:48:29: error: ‘BasicCommunication_DialNumber’ is not a member of ‘hmi_apis::FunctionID’
   event_engine::Event event(hmi_apis::FunctionID::BasicCommunication_DialNumber);
                             ^
make[2]: *** [src/components/application_manager/CMakeFiles/AMHMICommandsLibrary.dir/src/commands/hmi/dial_number_response.cc.o] Error 1
make[1]: *** [src/components/application_manager/CMakeFiles/AMHMICommandsLibrary.dir/all] Error 2
make: *** [all] Error 2
`

justinjdickow, 2016-07-13:
Looks like there was a previous PR #565 that fixed this issue. I'm not sure why it was closed.

sztankatt, 2016-07-13:
Hmm, not sure about that.

It seems that the whole `hmi_apis::FunctionID` cant be found, as there is no FunctionID defined in the HMI_API.xml file.

Maybe I am installing the wrong version... currently on the master branch.

jacobkeeler, 2017-04-19:
Closing as duplicate of #518

closed 2017-04-19 by jacobkeeler