smartdevicelink/sdl_core #682: RegisterAppInterface and DISALLOWED staus question

madhuy, 2016-07-08:
Does SDL core need to check IsApplicationWithSameAppIdRegistered()?

https://github.com/smartdevicelink/sdl_core/blob/master/src/components/application_manager/src/commands/mobile/register_app_interface_request.cc#L187

Here is use case(iOS):
1) launch the iOS app - it registers with SDL core
2) Kill the app in Mobile - Swipe out from Background
3) launch same app again - it try to register with SD core and Registerappinterface returns DISALLOWED. after this, all the RPC's are getting failed.

AnastasiyaBritanova, 2016-07-11:
@madhuy - yes, that is SDL requirement to return RAI (DISALLOWED) to the 2nd app if its appName is the same as another already registered app's appName.

In your scenario, it looks like SDL does not recognize the app is disconnected in step 2). 
Are you using app/SDL with heartbeat? 
The heartbeat functionality was introduced to resolve exactly the described issue.

madhuy, 2016-07-12:
@AnastasiyaBritanova I am using SDL core from https://github.com/smartdevicelink/sdl_core/releases/tag/4.0.12
iOS SDL proxy from https://github.com/smartdevicelink/sdl_ios/releases/tag/4.1.3

I am not sure this has Heartbeat functionality.

madhuy, 2016-07-13:
@AnastasiyaBritanova Please check this https://github.com/smartdevicelink/sdl_ios/pull/379/files
it looks like heartbeat functionality got deprecated in iOS

AnastasiyaBritanova, 2016-07-13:
@madhuy 
-> SDL4.0.12 supports heartBeat. 
-> please check smartDeviceLink.ini file (src/appMain/) -> [MAIN] -> `HeartBeatTimeout` must be non-zero to turn on HB on SDL.
-> mobile app in must also support HB.
-> HB is the feature of protocol v3: meaning, both SDL and app must negotiate (during startSession) they both support v3 protocol.
-> how HB works:
---- SDL sends HB_request and expects either HB_ACK or RPCs or data from mobile app. If nothing is received during timeout - SDL closes session with this app.
---- also the app sends HB_request and expects either HB_ACK or some RPCs from SDL. If nothing is received during timeout - the app is expected to close the session with SDL and start a new one.

In case the app **does not support HB**:
-> in your scenario, the app must manage to **correctly close its session with SDL** - otherwise, there are no other triggers for SDL to know the app is killed on the phone.

@justinjdickow please comment about the iOS app current behavior and whether HB is deprecated in it.

madhuy, 2016-08-05:
@AnastasiyaBritanova We tried by setting HeartBeat timeout value. the behavior is SDL core stopped communicating with Mobile apps. Attaching logs

[SDLCore_HeartBeat3_ios.log.zip]
(https://github.com/smartdevicelink/sdl_core/files/402989/SDLCore_HeartBeat3_ios.log.zip)

I think This could iOS SDL proxy may not support heartBeat.

@justinjdickow is HeartBeat deprecated in iOS SDL proxy ?

joeygrover, 2016-08-18:
@joeljfischer please look into this.

joeljfischer, 2016-08-18:
Yes, heartbeat is now deprecated in the iOS SDL Proxy as a result of unsolvable issues with running an app in the background being unable to use any sort of timers.

joeljfischer, 2016-08-18:
There may be a fix if the iOS library watches for a termination and attempts to close the session quickly before termination occurs.

joeljfischer, 2016-08-18:
@madhuy We'll help fix this in iOS SDK 4.3, but for now, in your app's `AppDelegate` `applicationWillTerminate:` method, send an `unregister` RPC Request. In the 4.3 SDK, we'll provide a method call that will do that for you.

madhuy, 2016-08-23:
@joeljfischer I tried your suggestion. by adding below code in AppDelegate applicationWillTerminate. It still not working.

  SDLUnregisterAppInterface *unregister = [[SDLUnregisterAppInterface alloc] init];
  [unregister setCorrelationID:[self getNextCorrelationId]];
  [self sendRPC:unregister];

joeljfischer, 2016-08-23:
Hmm... I'll double check when I get to the office tomorrow. I'm not seeing this issue when I do it that way.

kshala-ford, 2016-08-24:
Copied from https://developer.apple.com/library/ios/documentation/UIKit/Reference/UIApplicationDelegate_Protocol/ 

> applicationWillTerminate: (Called only when the app is running. This method is not called if the app is suspended.)

Thought core recognizes if remote peer on iap gets closed.

joeljfischer, 2016-08-24:
@kshala-ford This issue is related to when the app is force closed, which would call `applicationWillTerminate:`.

Are you referring to an iAP capability of recognizing when the remote peer gets closed?

joeygrover, 2016-08-29:
@madhuy are you seeing this the problem on Wifi or iAP connection?

madhuy, 2016-08-29:
@joeygrover iAP connection.

joeygrover, 2016-08-29:
In your Core implementation do you get any callback from the IAP api's that would indicate the IAP session has ended?

madhuy, 2016-08-29:
Let me communicate with our HU suppliers implementation. this is happening in only in HU's.

madhuy, 2016-09-19:
It is fixed

closed 2016-09-19 by madhuy

joeygrover, 2016-09-19:
Was this fixed via the Core 4.1.0 release or was it an implementation specific issue that was fixed?

madhuy, 2016-09-20:
it is implementation specific issue and it is fixed.