smartdevicelink/sdl_core #704: SDL.ini "LogsEnabled" Never Used

JackLivio, 2016-07-19:
[This setting in SmartDeviceLink.ini](https://github.com/smartdevicelink/sdl_core/blob/master/src/appMain/smartDeviceLink.ini#L26) has no effect on enabling/disabling logs.

LogsEnabled is never read by profile.cc or used anywhere in sdl core. The only way I have been able to disable logs is through the top level cmakelists or cmake -DENABLE_LOG=OFF.

jacobkeeler, 2016-08-05:
A way to implement this would be to include a new macro in logger.h, since it doesn't look like there are any log4cxx commands which are used directly without the use of some macro wrapper. Something like:

```cpp
#define DISABLE_LOGGER() ::log4cxx::LogManager::getLoggerRepository()->setThreshold(::log4cxx::Level::getOff())
```

The LogsEnabled option could then be implemented in profile.cc and a statement could be included after the config options are read in

```cpp
if (!profile::Profile::instance()->logs_enabled()) {
  DISABLE_LOGGER();
}
```

The only issue with this is that Core will still log out the configuration values while reading them in, but afterward will log nothing else.

I will submit a PR for this, but I'm not certain whether this would be a hotfix or a feature.

jacobkeeler, 2016-08-08:
This option is implemented already in the develop branch

closed 2016-08-08 by jacobkeeler