smartdevicelink/sdl_core #756: PerformInteraction should not require an empty array for interactionLayout KEYBOARD

asm09fsu, 2016-08-12:
Currently, we must pass an empty array for PerformInteraction's `interactionChoiceSetIDList` if we want to use the `interactionLayout` as `KEYBOARD`. Is there a reason for this?

This request body will work for PerformInteraction: 
```
{
    initialText = "Search";
    interactionChoiceSetIDList =     (
    );
    interactionLayout = KEYBOARD;
    interactionMode = "MANUAL_ONLY";
    timeout = 100000;
}
```

This will not:
```
{
    initialText = "Search for POI";
    interactionLayout = KEYBOARD;
    interactionMode = "MANUAL_ONLY";
    timeout = 100000;
}
```

TMelnyk, 2016-08-30:
Hello asm09fsu, 

Per the last actual version of MOBILE_API -> param _"interactionChoiceSetIDList"_ **_has NO "mandatory" field at all_**, it means that this param **is mandatory=true** **by default** 

>   <function name="PerformInteraction" functionID="PerformInteractionID" messagetype="request">
>     <description>Triggers an interaction (e.g. "Permit GPS?" - Yes, no, Always Allow).</description>
> 
> 
>     <param name=+"interactionChoiceSetIDList"+ type="Integer" minsize="0" maxsize="100" minvalue="0" maxvalue="2000000000" array="true">
>       <description>List of interaction choice set IDs to use with an interaction.</description>
>     </param>

As result, the **valid** request from mobile app will be:
_"PerformInteraction":_
"initialText" : string
"initialPrompt" : [strings]
**"interactionMode" : MANUAL_ONLY**
**"interactionChoiceSetIDList" : [ ]** _(param is empty BUT exists)_
"helpPrompt" : [strings]
"timeoutPrompt" : [strings]
"timeout" : int
"vrHelp" : [VRHelpItems]
**"interactionLayout" : KEYBOARD**

jacobkeeler, 2018-03-02:
It is unfortunate, but changing this would require a breaking change. As such, I'm closing this for now. If someone wants this to be changed, it will likely require a proposal.

closed 2018-03-02 by jacobkeeler