smartdevicelink/sdl_core #792: SDL must start <DefaultTimeout>+<RPCs_own_timeout> for all RPCs with own timer

TMelnyk, 2016-08-30:
_Issue:_
Currently SDL starts **only** _"RPCs_internal_timer"_ for waiting the response from HMI for all RPCs with own timer.
This _"RPCs_internal_timer"_ is spent also for transferring request to HMI, reading RPC, etc.
As result, some delays was observed at response from HMI side.

_Solution:_
SDL must start **DefaultTimeout** _(from .ini file)_ **+ RPCs_internal_timeout** for waiting the response from SDL side for **ALL** RPCs with own timer

**_List of impacted RPCs:_**
a. Alert _(“duration” param)_
b. PerformInteraction _(“timeout” param)_
c. Slider _(“timeout” param)_
d. ScrollableMessage _(“timeout” param)_

[image]

LitvinenkoIra, 2017-12-01:
Priority Medium.
Loss of functionality on SDL side with standard use-case.

LuxoftAKutsan, 2017-12-07:
Looks like a duplicate of #1871

jacobkeeler, 2018-03-08:
Changing labels to match duplicate issue #1871

GetmanetsIrina, 2018-05-03:
Script for reproducing issue: https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Defects/4_5/1880_DefaultAndRPCTimeoutGENERIC_ERROR.lua

LitvinenkoIra, 2018-11-14:
duplicate of https://github.com/smartdevicelink/sdl_core/issues/1880

JackLivio, 2019-10-16:
Fixed by #2495

closed 2019-10-16 by JackLivio