smartdevicelink/sdl_core #805: [SDLcore 4.1 release] iOS apps getting onHMIstatus NONE after 1 second of onHMIStatus FULL

madhuy, 2016-09-07:
1. start iOS app
2. Mobile app receives onHMIStatus FULL
3. after 1 sec Mobile app  receives onHMIstatus NONE.

my question do we need to change any thing is HS to compatible with SDLCore 4.1.0.

same works fine with android.

joeygrover, 2016-09-07:
@NicoleYarroch @joeljfischer can you take a look at this?

So Android apps do not receive the HMIStatus NONE after FULL?

madhuy, 2016-09-07:
@joeygrover Android app don't receive NONE after FULL

joeljfischer, 2016-09-07:
What version of sdl_ios, what version of iOS?

madhuy, 2016-09-07:
sdl_ios 4.2.2
iOS 9.3.2

more info, it works fine  if we launch app manually.
 it is not working, if app been launched by using onSystemRequest(LAUNCH_APP).

madhuy, 2016-09-12:
@joeljfischer any update on this?

BTPankow, 2016-09-12:
@madhuy I'm taking a look at this.  It seems as though the iOS app sends a OnHMIStatus notification with `hmiLevel = FULL` and  core responds with an `hmiLevel = NONE`.  This can be seen in the attached log excerpt ([OnHMIStatus.txt](https://github.com/smartdevicelink/sdl_core/files/468098/OnHMIStatus.txt)).  Could this be what is happening?

Would you be able to supply iOS logs that show the issue?

madhuy, 2016-09-12:
Attaching SDLcore and iOS logs
[ios_onHMIStatus_notworking.txt](https://github.com/smartdevicelink/sdl_core/files/468252/ios_onHMIStatus_notworking.txt)
[sdlcore_onHMIStatus_notwoking.txt](https://github.com/smartdevicelink/sdl_core/files/468253/sdlcore_onHMIStatus_notwoking.txt)

BTPankow, 2016-09-12:
@madhuy Thanks for the logs, it definitely looks like something strange is occurring.

What HMI are you using?  Does this happen with the [sdl_hmi](https://github.com/smartdevicelink/sdl_hmi)?

madhuy, 2016-09-12:
@BTPankow it is our own HMI. it was perfectly fine with 4.0 release . I am not sure why it is happening in 4.1 SDL core.

BTPankow, 2016-09-15:
I have a few more questions about the issue.  First, how is the app launched?  Is the HMI giving an explicit OnSystemRequest(LAUNCH_APP) or is it using the ActivateApp RPC (sequence diagram: [ActivateAppAppLaunch](https://d83tozu1c8tt6.cloudfront.net/media/builds/master/1/2a7691311b01e910c21316f915be72dfdc537733/sdl/activateapp/assets/ActivateAppAppLaunch.png))?  This could be something to check out, since it looks like the HMI is currently using the former and it may work better to use the latter, since just sending LAUNCH_APP would bypass some sdl_core processes.

Also, this issue currently looks like it may involve resumption, since the app is launched and then it seems like sdl_core tries to resume a previous state of hmi_level NONE when the app has already been given hmi_level FULL.
I haven’t been able to reproduce the issue, but I’m interested if this still occurs when you delete `app_info.dat` and `storage/` prior to running ./smartDeviceLinkCore.  This may prevent the current application from resuming to another state when launched with LAUNCH_APP.  Using ActivateApp may help fix this.

madhuy, 2016-09-15:
We are using OnSystemRequest(LAUNCH_APP) if the app is not registered with HMI.
if app registered with HMI we call ActivateApp.

we will check with your suggestion by deleting appinfo.dat and get back to you.

madhuy, 2016-09-15:
We deleted appinfo.dat and tried still not working.

attaching logs 
[sdlcore_ios_deleted_appinfo.txt](https://github.com/smartdevicelink/sdl_core/files/475584/sdlcore_ios_deleted_appinfo.txt)

madhuy, 2016-09-19:
I noticed this works, it does works some times. attaching working case logs 
[sdlCore_onHMIstatusWorking.txt](https://github.com/smartdevicelink/sdl_core/files/480736/sdlCore_onHMIstatusWorking.txt)

the difference between Working case and not working case 

Working case
-----------------
1. BasicCommunication.OnAppRegistered @ 12 Sep 2016 12:04:33,837
2. BasicCommunication.UpdateAppList with 2 apps @ 12 Sep 2016 12:04:33,837
3. **DEBUG [12 Sep 2016 12:04:33,941][StateControllerImpl] old: hmi_level 3, audio_state 2, system_context 0**

NOT working case
---------------------
1.  BasicCommunication.UpdateAppList with 1 app @ 12 Sep 2016 12:08:27,860
2. BasicCommunication.OnAppRegistered  @ 12 Sep 2016 12:08:27,865
3. **DEBUG [12 Sep 2016 12:08:28,055][StateControllerImpl] old: hmi_level 0, audio_state 2, system_context 0** 

I am not what cause the **old: hmi_level 0** to be 0. 

Please let me know if i can provide any info.

madhuy, 2016-09-19:
FYI. works fine with Wifi and SDL iOS proxy.

BTPankow, 2016-09-19:
Thanks for the logs and information.  As an update, I was able to reproduce this issue by activating the application from the HMI immediately after it is registered.
If there is a delay of about one second or more before sending ActivateApp, the issue does not appear and the iOS application does not receive an HMI status of NONE after FULL.
I'll try to take a closer look at what happens during this delay to see what could be the problem when there is no delay.

madhuy, 2016-09-20:
@BTPankow I did experiment by adding 2 sec delay to activate app after it is registered. it is working.

BTPankow, 2016-09-20:
I just submitted a pull request that should fix the issue.

It turns out that is was an issue with resumption like I had initially thought.  If the application is activated (and given HMI FULL) before it is resumed, the application will then receive an HMI NONE once resumption has finished.  PR #829 should fix this.

madhuy, 2016-09-20:
@BTPankow Thanks for quick fix. does resumption happens even if there is  no app_info.dat file ?

BTPankow, 2016-09-20:
@madhuy The `StartResumption` function will be called even if there was no preexisting app_info.dat file.  In most cases resumption will at least start (which had caused the apps to receive a new onHMIStatus notification).

madhuy, 2016-09-20:
@BTPankow your fix is working. Verified with our HMI.

madhuy, 2016-09-22:
@BTPankow This issue is reproducible even in 4.0 but frequency is very low and with different repro steps.

Repro Steps in 4.0
-------------------
1. SDL app is running in HU
2. Kill app ( swipe out from back ground in iOS)
3. Launch SDL app by using APP_LAUNCH 

we observe similar behavior. but frequency is very low. attaching logs 
[sdlcore_4.0_onHMIStatus_NONE.txt](https://github.com/smartdevicelink/sdl_core/files/488112/sdlcore_4.0_onHMIStatus_NONE.txt)

here is onHMIStatus call in logs

_21:12:34,426 OnHMIStatus(NONE)
21:12:34,605 OnHMIStatus(FULL)
21:12:37,563 OnHMIStatus(NONE)_

and also i see this message

_INFO  [31 Dec 2000 21:12:37,560][ResumeCtrl] Resume application after short IGN cycle_

BTPankow, 2016-09-22:
@madhuy Thanks for the logs and investigation.  I agree that this issue should be reproducible in earlier sdl versions.  This issue looks like it should occur when an application is activated immediately after it is registered with the HMI.

The fact that it is inconsistent likely has to do with timing and the speed with which sdl_core receives and processes the ActivateApp command.  However, I'll double check that this is the case.

BTPankow, 2016-09-23:
Submitted a second pull request (#840) that should fix this issue.  The other pull request (#829) may have had more side effects and may not have conformed to requirements.

Also, I checked the root cause and was able to reproduce the issue by just sending an ActivateApp from the HMI immediately after OnAppRegistered received.

joeygrover, 2016-10-12:
Pull request #840 was merged to fix this issue.

closed 2016-10-12 by joeygrover