smartdevicelink/sdl_core #809: log4cxx: Please initialize the log4cc system properly

willard1218, 2016-09-09:
[image]

error when I start ./smartDeviceLinkCode
How can I fix that?

BTPankow, 2016-09-09:
For the new master, I've found that the smartDeviceLinkCore executable should be run from inside the `build/bin/` directory, not `build/src/appMain/`.  I've submitted a pr to reflect this in the README.

jacobkeeler, 2016-09-09:
@willard1218 This is a result of the recently merged 4.1.0 release.

It is looking for the static log4cxx libraries which are put in the $SDL_BUILD/bin folder, the easiest way to get it working is to
1. Run `export LD_LIBRARY_PATH="."` (will only work for the current terminal session, you may want to put this in your `.bashrc` file)
2. As BTPankow said, start SDL Core from the $SDL_BUILD/bin folder rather than the $SDL_BUILD/src/appMain folder

@LuxoftAKutsan Is this the way that it is supposed to be set up, or is there supposed to be some way to get it running from src/appMain, as is currently in the README? These new steps will need to be added to the README in any case.

alrdguez, 2016-09-12:
Actually, running LD_LIBRARY_PATH=. ./smartDeviceLinkCore from bin directory doesn give initial error as @willard1218 posted but it throws another one related with log4cxx too:

```
$ LD_LIBRARY_PATH=. ./smartDeviceLinkCore
 log4cxx: setFile(SmartDeviceLinkCore.log,true) call failed.
log4cxx: IO Exception : status code = 13
log4cxx: setFile(ProtocolFordHandling.log,false) call failed.
log4cxx: IO Exception : status code = 13
log4cxx: setFile(TransportManager.log,false) call failed.
log4cxx: IO Exception : status code = 13
log4cxx: No output stream or file set for the appender named [SmartDeviceLinkCoreLogFile].
```

Any suggestions?

BTPankow, 2016-09-12:
@alrdguez This looks like an issue with permissions.  It seems as though log4cxx does not have permission to write the log files.

This is likely due to the fact that the log files are owned by root from previously running `./smartDeviceLinkCore` as root.  Running `ls -l *.log` should verify that the log files are owned by root.  Example:
```
-rw-r--r-- 1 root root    7658 Sep 12 16:17 ProtocolFordHandling.log
-rw-r--r-- 1 root root 1696490 Sep 12 16:17 SmartDeviceLinkCore.log
-rw-r--r-- 1 root root  156298 Sep 12 16:17 TransportManager.log
```

The easiest solution would be to delete the log files (plus other files owned by root) by running
```
rm -rf ProtocolFordHandling.log SmartDeviceLinkCore.log TransportManager.log app_info.dat storage
```

alrdguez, 2016-09-13:
Thanks a lot @BTPankow but it doesnt work, at least for me. I have used SDL since 3.8 but never had this issue. Hope 4.1.1 solves this.

sandeepf184, 2016-09-13:
I have the same issue

jacobkeeler, 2016-09-14:
@alrdguez @sandeepf184 This may be because you installed using sudo. You should only run `make install` rather than `sudo make install`, otherwise the `build/bin` directory will be owned by root.

Try deleting the `bin` directory and rerunning `make install` without sudo, and that should fix the issue. 

Alternatively, you could just change the owner of the folder to that of your user account using:
```
chown $YOUR_USER_NAME build/bin
```

alrdguez, 2016-09-15:
Thanks @jacobkeeler . That worked. :)

sandeepf184, 2016-09-15:
Thanks @jacobkeeler. For me export LD_LIBRARY_PATH="." worked.

willard1218, 2016-09-15:
Thanks @jacobkeeler . its worked.

jacobkeeler, 2016-09-23:
Closing with the merge of #810

closed 2016-09-23 by jacobkeeler