smartdevicelink/sdl_core #823: SendLocation uses Doubles instead of Floats

joeygrover, 2016-09-15:
The lat and long params are of type Double when they should be floats in the MOBILE_API for the struct Coordinate.

[See here](https://github.com/smartdevicelink/sdl_core/blob/master/src/components/interfaces/MOBILE_API.xml#L2293)

joeygrover, 2016-09-19:
[image] SendLocation uses doubles not floats</a>

sandeepf184, 2016-09-22:
Yeah Even I noticed that.

TMelnyk, 2016-09-28:
Hello @joeygrover 
1. "Double" type is **_correct type_** for _"latitudeDegrees"_ and _"longtitudeDegrees"_ params per existing requirements for SDL - as result, type will not be changed

2.  The mandatory of _"latitudeDegrees"_ and _"longtitudeDegrees"_ params of _SendLocation_ is planned to be changed from 'false' to 'true' in the next releases

jacobkeeler, 2016-09-28:
@TMelnyk Is there any reason why both of these parameters are "Float" type in [GPSData](https://github.com/smartdevicelink/sdl_core/blob/master/src/components/interfaces/MOBILE_API.xml#L1537-L1542)?

joeygrover, 2016-09-28:
I don't believe that's a correct requirement. We don't use doubles anywhere else in the Mobile_Api.

@stefanek22 @tpulatha Can you comment?

Also, changing from not required to required will introduce extra corner cases between versions and would require the RPC API version to be increased.

TMelnyk, 2016-09-29:
Hello @jacobkeeler , @joeygrover ,
Per discussion with Ford team the type of "_latitudeDegrees_" and "_lotitudeDegrees_" will be changed to **Float** at SendLocation_request and Coordinate struct
GPSData struct will remain without changes

jacobkeeler, 2016-10-04:
PR: #837

PR #837 (2016-10-17): Change Double type parameters in spec to Float type
PR Changes
  - Mobile API
    - longitudeDegrees and latitudeDegrees parameters in Coordinate and SendLocation changed to Float type from Double
  - HMI API
    - longitudeDegrees and latitudeDegrees parameters in Coordinate and GPSData changed to Float type from Double

Fixes #823
closed 2016-10-17 by jacobkeeler