smartdevicelink/sdl_core #831: [SDLcore 4.1 release] onPhoneCall notification is removed ?

madhuy, 2016-09-20:
if onPhoneCall is removed what is the best way to notify phone call status?

madhuy, 2016-09-20:
All these files are removed
on_emergency_event_notification.cc
on_play_tone_notification.cc
on_phone_call_notification.cc

joeygrover, 2016-09-21:
@AnastasiyaBritanova can you comment?

AnastasiyaBritanova, 2016-09-21:
@joeygrover yes, it was replaced with [OnEventChanged](https://github.com/smartdevicelink/sdl_core/blob/master/src/components/interfaces/HMI_API.xml#L2468)
It also replaces OnEmergencyEvent, OnDeactivateHMI - when sent with the appropriate eventTypes.

joeygrover, 2016-09-21:
[EventTypeEnums](https://github.com/smartdevicelink/sdl_core/blob/master/src/components/interfaces/HMI_API.xml#L1182)

@madhuy Does this satisfy your issue?

madhuy, 2016-09-21:
@joeygrover Could you please provide me updated sequence diagrams or updated SDL_integration guidlines?

joeygrover, 2016-09-21:
@AnastasiyaBritanova are there updated diagrams for this new flow?

madhuy, 2016-09-21:
@joeygrover @AnastasiyaBritanova Could you please let us know if any API's removed or added in 4.1 release. I know Playtone Notification removed and added the param("Playtone") in TTS.Speak .

AnastasiyaBritanova, 2016-09-22:
@madhuy @joeygrover 
1. the diagrams are attached
2. will prepare the list of changes and update you here
[image]
[image]
[image]

madhuy, 2016-09-26:
@AnastasiyaBritanova 

Could you please provide me list of changes in 4.1?

AnastasiyaBritanova, 2016-09-27:
hi @madhuy 
attached providing the diff of HMI_APIs between releases 4.0.12 and 4.1.0 (excel file).
Will get back a bit later with the list of related functional changes.

[HMI_API_diff_release4.0.12_vs_release4.1.0.xlsx](https://github.com/smartdevicelink/sdl_core/files/495541/HMI_API_diff_release4.0.12_vs_release4.1.0.xlsx)

closed 2016-10-03 by madhuy