smartdevicelink/sdl_core #838: Web SDL HMI doesn't follow the HMI flow

sandeepf184, 2016-09-22:
According to the HMI flow, when we send messages like dialnumber it should check for high priority events like Alert and Emergency Call etc. But I have seen in web hmi that when alert message is displayed and made a request for dialnumber , the dialnumber popup is displayed over alert message.  Can anyone please check with it and help me with that.

JackLivio, 2016-09-23:
@sandeepf184 Here's a fix
https://github.com/smartdevicelink/sdl_hmi/pull/5

joeygrover, 2016-10-31:
Can you reference which guide you found this information?

sandeepf184, 2016-10-31:
https://smartdevicelink.com/en/docs/hmi/master/basiccommunication/dialnumber/

Refer the sequence diagrams

jacobkeeler, 2017-04-18:
Closing, issue moved to sdl_hmi repository (smartdevicelink/sdl_hmi#18)

closed 2017-04-18 by jacobkeeler