smartdevicelink/sdl_core #860: The reference of wild pointer cause to memory leak

wzhhnet, 2016-09-30:
https://github.com/smartdevicelink/sdl_core/blob/master/src/components/dbus/src/dbus_adapter.cc
`bool DBusAdapter::SetValue(...
...
    case ford_message_descriptions::ParameterType::String:
      type = DBUS_TYPE_STRING;
      stringValue = param.asString().c_str();
`
temporary string object was returned, but it was not copied, it's internal pointer member was copied by stringValue. after this, the temporary string object was destroyed and stringValue become a wild pointer.

chenbd, 2017-04-11:
is this a problem which param.asString() return a temp string, after c_str() call, pointer is returned from temp string after temp string is destructed?
which means use a pointer from string.c_str() after string is destructed.

LitvinenkoIra, 2017-12-01:
Priority Medium.
Related to memory leak, but there is no defined use-case and behavior.

jacobkeeler, 2018-03-06:
Fixed with the merge of #2051

PR #2051 (2018-03-06): Fix wild pointer in dbus adapter
Fixes #860 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Summary
Fix wild pointer caused by the destruction of a temporary string before use

### Changelog
##### Bug Fixes
* Fix wild pointer

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
closed 2018-03-06 by jacobkeeler