smartdevicelink/sdl_core #885: For some RPCs, Core does not unsubscribe from softbuttons after receiving a response

jacobkeeler, 2016-10-07:
For all of the popup-based RPCs which contain softButtons, Core is supposed to stop processing softButton events once a response is received for the RPC. Currently the only one that does this is [ScrollableMessage](https://github.com/smartdevicelink/sdl_core/blob/master/src/components/application_manager/src/commands/mobile/scrollable_message_response.cc#L55), for all of the others Core stays subscribed to their softButtons after receiving a response.

Affected RPCs: Alert, ShowConstantTBT, AlertManeuver, UpdateTurnList

jacobkeeler, 2016-10-24:
[image] [Alert & Show Constant TBT & Alert Maneuver & Update Turn List Do not unsubscribe after conclusion of pop up.](https://trello.com/c/iNwelphz/111-alert-show-constant-tbt-alert-maneuver-update-turn-list-do-not-unsubscribe-after-conclusion-of-pop-up)

LitvinenkoIra, 2017-12-01:
Priority Low.
Low affecting for functionality.

E-SAITO-TMC, 2020-04-14:
@joeygrover -san
A fix has been proposed for this issue. When will it be merged? For the next HU development of TOYOTA, it is necessary to improve the quality of SDLCore, so we believe it is necessary to modify this issue.

theresalech, 2020-04-15:
Hi @E-SAITO-TMC, contents for SDL releases are determined by the Steering Committee, and the Project Maintainer works on features and bug fixes raised as high priority by Steering Committee representatives.  This particular issue has not yet been raised as high priority.  However, if Toyota would like to see this fix in the next release of SDL Core (planned for October 28, 2020), we can make note of such.  

Please confirm if this is priority for Toyota.  If so, we will include this issue in our proposed release contents, which will be presented to the Steering Committee and voted upon/finalized in June.  Thank you!

theresalech, 2020-04-20:
@E-SAITO-TMC, thank you. We have documented that this is medium priority for Toyota, and will include for Steering Committee review and vote when planning the contents for the Q4 Core release. This planning will take place in June.

JackLivio, 2020-06-25:
@jacobkeeler

JackLivio, 2020-06-25:
@JackLivio

ShobhitAd, 2021-02-02:
Closed via https://github.com/smartdevicelink/sdl_core/pull/2629

PR #2629 (2021-02-02): Fix SDL does not unsubscribe from softbuttons after receiving a response.
Fixes #885.

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Covered by Unit tests.

### Summary
When SDL recives AlertManeuverResponse, AlertResponse, UpdateTurnListResponse, ScrollableMessageResponse, ShowConstantTBTResponse SDL unsubscribes from soft 
buttons before sends response to mobile.
Added unit test cases for specified responses.

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewfo [... cut]
closed 2021-02-02 by ShobhitAd