smartdevicelink/sdl_core #890: Build fails if -DEXTENDED_MEDIA_MODE=ON

yang1070, 2016-10-11:
[ 95%] Building CXX object src/components/media_manager/CMakeFiles/MediaManager.dir/src/media_manager_impl.cc.o
In file included from /home/zhimin/github/sdl_core/src/components/media_manager/src/media_manager_impl.cc:45:0:
/home/zhimin/github/sdl_core/src/components/media_manager/include/media_manager/audio/a2dp_source_player_adapter.h: In member function ‘void media_manager::A2DPSourcePlayerAdapter::set_connection_handler(connection_handler::ConnectionHandlerImpl*)’:
/home/zhimin/github/sdl_core/src/components/media_manager/include/media_manager/audio/a2dp_source_player_adapter.h:63:5: error: ‘connection_handler_’ was not declared in this scope
     connection_handler_ = connection_handler;
     ^
src/components/media_manager/CMakeFiles/MediaManager.dir/build.make:158: recipe for target 'src/components/media_manager/CMakeFiles/MediaManager.dir/src/media_manager_impl.cc.o' failed
make[2]: *** [src/components/media_manager/CMakeFiles/MediaManager.dir/src/media_manager_impl.cc.o] Error 1
CMakeFiles/Makefile2:2429: recipe for target 'src/components/media_manager/CMakeFiles/MediaManager.dir/all' failed
make[1]: *** [src/components/media_manager/CMakeFiles/MediaManager.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

this happens in release/4.2.0, and mater with tag 4.1.0, 4.1.1. Didn't try other tags.

jacobkeeler, 2016-10-11:
@yang1070 Working on this, I believe I have fixed all of the issues with the build and will likely be submitting a PR for it later this week

yang1070, 2016-10-11:
@jacobkeeler  How do you set up the GStreamer? I mean the include path and link libs. I changed the code locally to remove the connection_handler error. Then I met a "utils/logger.h" file cannot find error. I think somehow the FindGstreamer-1.0.cmake file in the project should be called to setup the Macro, but I didn't figure out how.

jacobkeeler, 2016-10-12:
@yang1070 The connection_handler issue should have been fixed with the merge of #880.

I went through all of the build issues and fixed them in #900, so the changes in that PR should fix your issues. Note that you will need to manually specify the gstreamer directory as well:

```
export GSTREAMER_DIR=$PATH_TO_GSTREAMER
```

yang1070, 2016-10-13:
@jacobkeeler Thank you very much for the quick response. The pull request #880 and #900 together do fix the build and core crash issue. 
however, the html hmi associated with the repository has anthoer issue https://github.com/smartdevicelink/sdl_hmi/issues/8

jacobkeeler, 2016-10-24:
Fixed with #900

closed 2016-10-24 by jacobkeeler