smartdevicelink/sdl_core #905: SDL core 4.1 - policy.sqlite file crated in current directory instead of 'AppStorageFolder'

madhuy, 2016-10-13:
in 4.0, it was creating in 'AppStorageFolder' path
in 4.1 it is creating in current directory 

Looked at the code,
can we add below code [here](https://github.com/smartdevicelink/sdl_core/blob/master/src/components/policy/src/sql_pt_representation.cc#L322)

`#ifndef __QNX__
  std::string path = get_settings().app_storage_folder();
  if (!path.empty()) {
    db_->set_path(path + "/");
  }
#endif // __QNX__`

jacobkeeler, 2016-10-14:
It looks like the change was made in this commit: https://github.com/smartdevicelink/sdl_core/commit/96f5dd24e02a6f769248c4c6b2db055fca82cadd for #477.

@Kozoriz, @LuxoftAKutsan, or @AByzhynar is there any reason that this path was changed?

LuxoftAKutsan, 2016-10-18:
@jacobkeeler I agree, seems like it was deleted my mistake. By note, that there is redundant ifdefs that probably should be removed ( __QNX__), in other case this code will not work on QNX.

jacobkeeler, 2016-10-18:
@LuxoftAKutsan So, the `ifndef __QNX__` statement should not be added back in then?

I'll make a PR with these changes...

LuxoftAKutsan, 2016-10-18:
@jacobkeeler right

jacobkeeler, 2016-10-19:
PR: #908

PR #908 (2016-10-26): Set policy db path to storage folder
Quick change which sets the path for `policy.sqlite` to the app storage folder defined in smartDeviceLink.ini

Fixes #905
closed 2016-10-26 by jacobkeeler