smartdevicelink/sdl_core #906: Investigation on 'year 2038 integer overflow' issue

madhuy, 2016-10-13:


joeygrover, 2016-10-14:
Can you please include your concerns and what you think this might affect in the description?

madhuy, 2016-10-14:
@joeygrover We need support HU software until 2050. our partner asked us to investigate impact of integer over flow issue.
Does SDL core uses dates or times for either computation or diagnostic logging ? if Yes, How does it impacts the overall SDL functionality?

JackLivio, 2016-10-19:
@madhuy We set our time to the year 2050 and first glance nothing seemed to be broken. Have you tried doing this on your system?

madhuy, 2016-10-20:
@JackLivio 

better way to test this is 

1. Change Time to "03:13:00 UTC on Tuesday, 19 January 2038"
2. start SDL core
3. wait until "03:14:07 UTC on Tuesday, 19 January 2038 " , here it will overflows. 
4. Check behavior of SDL core

More [Info](https://en.wikipedia.org/wiki/Year_2038_problem)

joeygrover, 2016-10-20:
@madhuy have you done this test? What results did you have?

madhuy, 2016-11-14:
We did experiment with above steps in our environment , we don't see any issues. closing this issue

closed 2016-11-14 by madhuy