smartdevicelink/sdl_core #913: release/4.2.0 coredump when SyncProxyTester was trying to connect from wifi

yang1070, 2016-10-21:
zhimin@SYNC3:~/github/sdl/bin$  ./smartDeviceLinkCore 
accept_buf: yfQA9drYMhYefsSbledbcH21eI8=
accept_buf: buzEt3UgddaH3VN/lu0SsO60gVU=
accept_buf: J41AtZuOVFrmnr4Yiv/AeM4PKI8=
accept_buf: d3D8XKuIT52ncEpTy7pLaV1rJ+w=
accept_buf: YsxVNyySNxbTejiU7+RsWxfYk/Y=
accept_buf: iQVr7UD05PedsiHqMnIyudcg00c=
accept_buf: xXT5rD1syWNwxAosQ20sI4pSqKU=
ERROR!!! /home/zhimin/github/sdl_core/src/3rd_party-static/MessageBroker/src/server/mb_tcpserver.cpp:96 Received 4096 actual data length 4828
smartDeviceLinkCore: /home/zhimin/github/sdl_core/src/components/application_manager/src/usage_statistics.cc:115: void application_manager::UsageStatistics::RecordHmiStateChanged(mobile_apis::HMILevel::eType): Assertion `(!"Unreachable code")' failed.
ERROR!!! /home/zhimin/github/sdl_core/src/3rd_party-static/MessageBroker/src/server/mb_tcpserver.cpp:97 Incomplete messageAborted (core dumped)

JackLivio, 2016-10-24:
@yang1070 What HMI are you connecting to SDL Core?

MaiAnhBao, 2016-11-10:
passport@userubuntu:~/sdl/build/bin$ LD_LIBRARY_PATH=. ./smartDeviceLinkCore 
accept_buf: IkD1cBMU3xynaHYEXCRSEImgJc4=
accept_buf: fKykd5jAhWbDPEC/xeurtBXHmRc=
accept_buf: LqbKTByCzLDNEZir3BWR+r6mX1w=
accept_buf: JaaYWswgzw3VYhA8MMh7UEytKUc=
accept_buf: pf22UAkKWnMHbyafuVChB5WVmAw=
accept_buf: RNBKJ2U+zK+EHa1hEtlWfSRbfeY=
accept_buf: IFHan8/dB+iRDO0Z7Rtwv2M8L2I=
accept_buf: ITtZ1faZrll/DCLUPYe54EdH628=
ERROR!!! /home/passport/sdl/sdl_core/src/3rd_party-static/MessageBroker/src/server/mb_tcpserver.cpp:96 Received 4096 actual data length 4916
ERROR!!! /home/passport/sdl/sdl_core/src/3rd_party-static/MessageBroker/src/server/mb_tcpserver.cpp:97 Incomplete messageERROR!!! /home/passport/sdl/sdl_core/src/3rd_party-static/MessageBroker/src/server/mb_tcpserver.cpp:96 Received 4096 actual data length 4916

I have the same problem. In fact, I use HMI from https://github.com/smartdevicelink/sdl_hmi.git. Note that I try to change port in HMI from 8087 to 8086 as the NSMessageBroker file.

roberthartman, 2017-03-27:
I see the same error as @MaiAnhBao. Is there a solution?

jacobkeeler, 2017-04-18:
@MaiAnhBao @roberthartman The MAX_RECV_DATA constant can be changed in the message broker to a value larger than 4916 in order to get rid of this error message. However, it should not actually cause any issues to leave it as is, everything appears to work fine even with the error message.

closed 2017-04-18 by jacobkeeler

cat-fang, 2018-12-01:
How did you solved the problem? Can you share the solution? thanks very much.