smartdevicelink/sdl_core #932: [GENIVI] [Policy] "EXTERNAL_PROPRIETARY" flow: appID Management

TMelnyk, 2016-11-03:


DrachenkoAnastasiia, 2016-11-21:
_**Description:**_
Policies code at open source SDL should be the same as at Ford-Specific branch.
To "switch on" the extended Policies -> SDL should be built with ***-DEXTENDED_POLICY: EXTERNAL_PROPRIETARY*** flag
. 
***Impact:***
SDL 
HTML5 HMI 
SPT 
TestCases 
Requirements

DrachenkoAnastasiia, 2016-11-22:
:one: _In case_ the application with "appName" is successfully registered to SDL
**and** the updated policies do not have this "appName" in this app's specific policies (= "nicknames" filed in "\<appID\>" section of "app_policies")
SDL must:
send OnAppInterfaceUnregistered (APP_UNAUTHORIZED) to such application.
:information_source: **_Information:_**
a. The application itself is responsible for closing session with SDL
b. The next registration request from such app will be disallowed by SDL 

:two: _In case_ the application sends RegisterAppInterface request with 
\> the "appName" value that is listed in this app's specific policies
\> other valid parameters
SDL _must:_
successfully register such application: RegisterAppInterface_response (\<applicable resultCode\>,  **success:true**)
:information_source: **_Information:_**
\<applicable resultCode\> is the one that applies to successful registration:
WRONG_LANGUAGE
RESUME_FAILED
WARNINGS
SUCCESS

:three: _In case_ the application sends RegisterAppInterface request with the "appName" value that 
\> is not listed in this app's specific policies
\> is the same as another already-registered application has
SDL _must:_
return RegisterAppInterface_response (DISALLOWED, success: false)

:four:  PoliciesManager **must disallow** the app`s registration IN CASE the app`s nickname does not match those listed in Policy Table under the appID this app registers with.

:five: SDL must be **case-insensitive** when comparing the value of _"appName"_ received within _RegisterAppInterface_ against the value(s) of _"nicknames"_ section for the corresponding appID provided by the application.

:six: _In case_ **app_specific** policies are assigned to app
**AND** this app sends _ChangeRegistration_ request with "appName" that **does not exist in "nicknames"** field in PolicyTable
SDL _must_
respond with (DISALLOWED, success:false) to this application _(not unregister it)_.

jacobkeeler, 2017-04-18:
Closing due to being a requirement rather than an issue

closed 2017-04-18 by jacobkeeler