smartdevicelink/sdl_core #933: [GENIVI] [Policy] "EXTERNAL_PROPRIETARY" flow: Policy Table Update

TMelnyk, 2016-11-03:


DrachenkoAnastasiia, 2016-11-21:
***Description:***
High-level idea:
Policies code at open source SDL should be the same as at Ford-Specific branch.
To "switch on" the extended Policies -> SDL should be built with ***-DEXTENDED_POLICY: EXTERNAL_PROPRIETARY*** flag

***Impact:***
SDL 
HTML5 HMI 
SPT 
TestCases 
Requirements

KhrystynaDubovyk, 2016-11-22:
Description:
 High-level idea:
 To "switch on" the extended Policies -> SDL should be built with -DEXTENDED_POLICY: EXTERNAL_PROPRIETARY flag

Impact:
 SDL
 HTML5 HMI
 SPT
 TestCases
 Requirements

## 1. When to Request an Update to the Policy Table
**1.1. The triggers of PTU**
* The policies manager must request an update to its local policy table when an appID of a registered app is not listed on the Local Policy Table and the device the application is running on **is consented**. 
* PoliciesManager must check the stored status of PTUpdate upon every Ign_On and in case it is _UPDATE_NEEDED_, PoliciesManager **must** initiate the PTUpdate sequence by sending SDL.PolicyUpdate request to HMI right after the first application has registered on SDL.
* PoliciesManager **must** initiate the PT Update in case the app that has never received the updated policies registers from non-consented device and then the User consents this device.
**Note:**
This requirement applicable ONLY for EXTENDED_PROPRIETARY flow
* PoliciesManager must initiate the PT Update through the app from consented device IN CASE the app that has never received the updated policies registers from non-consented device and then the User does NOT consent this device (device should not be used e.i. no second query for user consent should be sent to HMI).
**Note:** SDL initiates PTU for application on unconsented device only in case the application has been selected in the app list on HMI and HMI notifies SDL about via _SDL.ActivateApp(isSDLAllowed:false)_
* In case SDL receives SDL.OnPolicyUpdate notification from HMI, PoliciesManager must start the procedure of Policy Table Update (that is, create PT Snapshot, send it to HMI for encryption, and etc. what is defined by related requirements) 
**HMI Important information:** HMI should send **SDL.OnPolicyUpdate** in case the new PT Snapshot is required from PoliciesManager . Currently **this notification is not used in any HMI workflows**, but in case of necessity, HMI may use OnPolicyUpdate for getting the newest Snapshot (for example, during retry sequence or by user request).
* PoliciesManager **must** initiate the PT Update sequence (that is, PT Exchange) upon User`s request delivered to SDL via SDL.UpdateSDL() from HMI and provide a response on a request with current PTU status to HMI.
_**Information:**_ The possibility for the User to request a PT update is dependent on the HMI of the system.
* In case **the amount of ignition cycles** notified by HMI via BasicCommunication.OnIgnitionCycleOver **gets equal** to the value of **"exchange_after_x_ignition_cycles"** field ("module_config" section) of policies database, SDL must trigger a PolicyTableUpdate sequence
_Example of PolicyTable value:_
```
{
  "policy_table": {
    "module_config": {
      "exchange_after_x_ignition_cycles": 100,
... }
```
* If SDL gets OnVehcileData ("odometer") notification from HMI and the difference between between **current "odometer" value_2** and **"odometer" value_1** when the previous UpdatedPollicyTable was applied is equal or greater than to the value of "exchange_after_x_kilometers" field ("module_config" section) of policies database, the policies manager **must** request an update to its local policy table after "N" kilometers.
**Note:** "N" shall be defined within the Local Policy Table under exchange_after_x_kilometers section.
_Example of PolicyTable value:_
```
 {
  "policy_table": {
    "module_config": {
      "exchange_after_x_kilometers": 100,
... }
```
* The policies manager **must** request an update to its local policy table after "N" days only if the system provided time is available.
**Note:** "N" shall be defined within the Local Policy Table under _exchange_after_x_days_ section.

jacobkeeler, 2017-04-18:
Closing due to being a requirement rather than an issue

closed 2017-04-18 by jacobkeeler