smartdevicelink/sdl_core #934: [GENIVI] [Policy] "EXTERNAL_PROPRIETARY" flow: App Permissions

TMelnyk, 2016-11-03:


DrachenkoAnastasiia, 2016-11-21:
_**Description:**_
High-level idea:
Policies code at open source SDL should be the same as at Ford-Specific branch.

To "switch on" the extended Policies -> SDL should be built with ***-DEXTENDED_POLICY: EXTERNAL_PROPRIETARY*** flag

_**Impact:**_
SDL 
HTML5 HMI 
SPT 
TestCases 
Requirements

JackLivio, 2017-02-20:
From the document [Diff_between_4_2_1_and_4_3_0.pdf](https://github.com/smartdevicelink/sdl_core/files/771065/Diff_between_4_2_1_and_4_3_0.pdf)

```
Functionality added:
After successful registration and data consent, in case of app activation user may be prompted to allow/disallow additional functions for specific apps, for example:
"DrivingCharacteristics"
"Notifications"
"Location" "VehicleInfo"

```

AStasiuk, 2017-02-20:
Is there anything that is not clear or do need any help?

JackLivio, 2017-02-20:
@AStasiuk I was just posting that description since there was none listed for this github issue. Is there anything more to this CRQ that you would add the describes the added app permissions functionality?

AStasiuk, 2017-02-20:
@JackLivio for this CRQ exist set of Functional and Non-Functional Requirements:
### 
Functional Requirements set:
 
 Notify application about permissions;
 Validate the RPC request;
 Validate the notifications;
 No user notification on app registering;
 HMI notifying about the permissions;

### Non-Functional Requirements set:

 Mobile_API: OnPermissionsChange notification;
 HMI_API: SDL.OnAppPermissionChanged notification;

Each of described above requirements set could be detailed to one atomic.
Please let me know if this info should be added in this PR.

jacobkeeler, 2017-04-18:
Closing due to being a requirement rather than an issue

closed 2017-04-18 by jacobkeeler