smartdevicelink/sdl_core #935: [GENIVI] [Policy] "EXTERNAL_PROPRIETARY" flow: General SDL behavior

TMelnyk, 2016-11-03:


DrachenkoAnastasiia, 2016-11-21:
_**Description:**_
High-level idea:
To "switch on" the extended Policies -> SDL should be built with _**-DEXTENDED_POLICY: EXTERNAL_PROPRIETARY**_ flag

_**Impact:**_
SDL 
HTML5 HMI 
SPT 
TestCases 
Requirements

KhrystynaDubovyk, 2016-11-22:
**Description:**
 High-level idea:
 To "switch on" the extended Policies -> SDL should be built with _**-DEXTENDED_POLICY: EXTERNAL_PROPRIETARY**_ flag

### **General non-functional requirements**
In preparation for a Policy Table Exchange, Policies Manager must have enough memory to hold a Policy Table Snapshot in memory.

**Information:** Policy Tables (Local, Snapshot, and Update) have an artificial upper file size limit of 1 Mbyte. 
 (Currently the Ford Backend ultimately defines the upper limit of the size of the payload of a syncP packet as 1 MB. Currently, if the backend received a packet over 1 MB, it would be rejected. However, we could increase the limit on the backend in the future if needed, so I don't think we should put any hard limit in the Policies Manager.)
Communication of Policy manager and mobile device must not make discernible difference in system operation.

_**Impact:**_
 SDL
 HTML5 HMI
 SPT
 TestCases
 Requirements

jacobkeeler, 2017-04-18:
Closing due to being a requirement rather than an issue

closed 2017-04-18 by jacobkeeler