smartdevicelink/sdl_core #936: [GENIVI] [Policy] "EXTERNAL_PROPRIETARY" flow: Related HMI API

TMelnyk, 2016-11-03:


DrachenkoAnastasiia, 2016-11-21:
_**Description:**_
High-level idea:
To "switch on" the extended Policies -> SDL should be built with **-DEXTENDED_POLICY: EXTERNAL_PROPRIETARY*** flag

_**Impact:**_ 
SDL 
HTML5 HMI 
SPT 
TestCases 
Requirements

JackLivio, 2017-02-20:
From the document [Diff_between_4_2_1_and_4_3_0.pdf](https://github.com/smartdevicelink/sdl_core/files/771065/Diff_between_4_2_1_and_4_3_0.pdf)

Updates in HMI_API

1. SDL.OnAllowSDLFunctionality
```
<function name="OnAllowSDLFunctionality" messagetype="notification">
     <description>Initiated by HMI. Notifies about user/HMI allowing SDL
functionality or disallowing access to all mobile apps. Needed if HMI has additional
ways of asking user about this (i.e. Settings etc)</description>
     <param name="device" type="Common.DeviceInfo" mandatory="false">
        <description>If no device is specified permission counts for SDL
functionality in general.</description>
</param>
  <param name="allowed" type="Boolean" mandatory="true">
    <description>Must be true if allowed</description>
  </param>
  <param name="source" type="Common.ConsentSource" mandatory="true"/>
</function>
<struct name="DeviceInfo">
  <param name="name" type="String" mandatory="true">
    <description>The name of the device connected.</description>
</param>
  <param name="id" type="String" mandatory="true">
    <description>The ID of the device connectedi: either hash of device's USB serial
number(in case of USB connection) or has of device's MAC address(in case of
BlueTooth or WIFI connection</description>
  </param>
  <param name="transportType" type="Common.TransportType" mandatory="false">
    <description>The transport type the named-app's-device is connected over
HU(BlueTooth, USB or WiFi). It must be provided in OnAppRegistered and in
UpdateDeviceList</description>
</param>
  <param name="isSDLAllowed" type="Boolean" mandatory="false">
    <description>Sent by SDL in UpdateDeviceList. ’true’ – if device is allowed for
PolicyTable Exchange; ‘false’ – if device is NOT allowed for PolicyTable Exchange
</description>
  </param>
</struct>
<enum name="ConsentSource">
    <element name="GUI"/>
    <element name="VUI" />
</enum>

```
2. SDL.OnAppPermissionConsent
```
<function name="OnAppPermissionConsent" messagetype="notification">
      <description>Initiated by HMI for specifying the allowance for the application
to perform some functionality. Duplicates functionality of the request, needed if
HMI has specific ways to allow/disallow functionality (i.e.Setting
Menu)</description>
      <param name="appID" type="Integer" mandatory="false">
          <description>Information about the application. See HMIApplication. If
omitted - allow/disallow all applications </description>
      </param>
      <param name="consentedFunctions" type="Common.PermissionItem"
mandatory="false" array="true" minsize="1" maxsize="100">
      </param>
      <param name="source" type="Common.ConsentSource" mandatory="true"/>
     </function>
<struct name="PermissionItem">
<param name="name" type="String" mandatory="true">
<description>Code of message of user-friendly text about functional group to be
allowed/disallowed</description>
</param>
<param name="id" type="Integer" mandatory="true">
<description>Id of corresponding functional group, assigned by SDL.</description>
</param>
<param name="allowed" type="Boolean" mandatory="false">
<description>Specifies whether functionality was allowed/disallowed. If ommited - no
information about User Consent is yet found for app.</description>
</param>
 
</struct>
<enum name="ConsentSource">
    <element name="GUI"/>
    <element name="VUI" />
</enum>
```
3. SDL.GetListOfPermissions
```
<function name="GetListOfPermissions"  messagetype="request">
      <description>Request from HMI to SDL to get list of permissions for app. If no
app specified - list of permissions for all apps.</description>
      <param name="appID" type="Integer" mandatory="false"/>
</function>
<function name="GetListOfPermissions" messagetype="response">
      <param name="allowedFunctions" type="Common.PermissionItem" mandatory="true"
array="true" minsize="0" maxsize="100">
        <description>If no permissions were specified for application the array will
come empty.</description>
      </param>
</function>
 
```

jacobkeeler, 2017-04-18:
Closing due to being a requirement rather than an issue

closed 2017-04-18 by jacobkeeler