smartdevicelink/sdl_core #937: [GENIVI] [Policy] "EXTERNAL_PROPRIETARY" flow: Policies Security

TMelnyk, 2016-11-03:


DrachenkoAnastasiia, 2016-11-21:
_**Description:**_
To "switch on" the extended Policies -> SDL should be built with ***-DEXTENDED_POLICY: EXTERNAL_PROPRIETARY*** flag

_**Impact:**_
SDL 
HTML5 HMI 
SPT 
TestCases 
Requirements

KhrystynaDubovyk, 2016-11-22:
_**Description:**_
To "switch on" the extended Policies -> SDL should be built with ***-DEXTENDED_POLICY: EXTERNAL_PROPRIETARY*** flag
### **Functional requirments**

1. The format of certificatefrom _“module_config”_ section of PolicyTable **must be changed** from PKCS12 to **PEM**.
_**Information:**_ 
The PEM format **must allow** the chain of certificates + private key to be bundled together.
OpenSSL which performs the handshake **will fail** the TLS handshake in case the certificate from PT **does NOT satisfy** the standard/encryption/format.  

2. In case the 'certificate' field 
 -> of 'module_config' section 
 -> of policies database 
**is empty or invalid or expired**, SDL must fail the TLS handshake 

3. In case during first ignition cycle PolicyTable starts uploading certificates **from** 
 -> 'certificate' field 
 -> of 'module_config' section 
 -> of PreloadedPT 
 -> to the policies database
**AND** the 'certificate' field _**is empty**_ SDL must **continue working** as assigned (meaning, this is not a reason for SDL to fail start up).

4. During TLS handshake performed while opening SecureService, SDL's SecurityManager must **send the certificate from "module_config" section of PolicyTable** to mobile app.

 _**Information:**_ 
 In case **TLS handshake fails**, SDL must behave depending on \<service\> requested for encryption:
**a.** in case \<service\> is a value of _"ForceProtectedService"_ param of .ini file, this \<service\> must receive **NACK** from SDL (\<service\> must not be started as non-encrypted).
**b.** in case \<service\> is **not** a value either of _"ForceProtectedService"_  or _"ForceUnprotectedService"_ params of .ini file, this \<service\>must receive **ACK with encryption flag OFF** (\<service\> must be started as non-encrypted).
**Note:** _for Gen3 (Panasonic) implementation for StartServices **10 and 11** mobile app must receive StartService(**NACK**) from SDL (that is, must be listed as values of _"ForceProtectedService"_ param of .ini file)_
_**Additional information:**_
 -> App is expected to validate this (module's = HeadUnit's) certificate to know whether it's signed by Ford and not expired. 
 -> In case HeadUnit's certificate is not signed by Ford or is expired, app side is expected to fail the TLS handshake (meaning: the SecureService will not be opened).

**Note:** _"certificate" correspondence to this format is not the case for PolicyTable validation (meaning, if the "certificate" string does not contain any 'begin' / 'end' / etc - PolicyTable is still valid)._ 
```
-----BEGIN RSA PRIVATE KEY-----
{PRIVATE_KEY_DATA_ENCODED_BASE64}
-----END RSA PRIVATE KEY-----
-----BEGIN CERTIFICATE-----
{CERTIFICATE_DATA_ENCODED_BASE64}
-----END CERTIFICATE-----
-----BEGIN CERTIFICATE-----
{CERTIFICATE_DATA_ENCODED_BASE64}
-----END CERTIFICATE-----
-----BEGIN CERTIFICATE-----
{CERTIFICATE_DATA_ENCODED_BASE64}
-----END CERTIFICATE-----
```
 -> Certificate standard is X.509
 -> Certificate format is **PEM** (which allows **multiple certificates + private key** to be bundled together)
 -> OpenSSL which performs the handshake will fail the TLS handshake in case the certificate from PT does not satisfy the above standard/encryption/format.

_**Impact:**_
SDL 
HTML5 HMI 
SPT 
TestCases 
Requirements

JackLivio, 2017-02-09:
@KhrystynaDubovyk 

Regarding 
```
-> App is expected to validate this (module's = HeadUnit's) certificate to know whether it's signed by Ford and not expired.
-> In case HeadUnit's certificate is not signed by Ford or is expired, app side is expected to fail the TLS handshake (meaning: the SecureService will not be opened).
```

Will that effect non Ford implementations of SDL? Will other integrators be able to support this feature with their own certificates or is this feature specific to Fords security?

TMelnyk, 2017-02-13:
@KhrystynaDubovyk , please remove the following Note:

> Note: for Gen3 (Panasonic) implementation for StartServices 10 and 11 mobile app must receive StartService(NACK) from SDL (that is, must be listed as values of "ForceProtectedService" param of .ini [...]

This Note relates to F-S only (SYNC3 notes)

TMelnyk, 2017-02-13:
@JackLivio , note related to SYNC3 Pana_Ford-specific will be removed from this PR
Actually each integrator must have own certificate

All req-s and info related to TLS Handshake _(=certificate exchange)_ should be _(and will be)_ described in Security feature at github

jacobkeeler, 2017-04-18:
Closing due to being a requirement rather than an issue

closed 2017-04-18 by jacobkeeler