smartdevicelink/sdl_core #1003: [SDL4.0]SDL does not create icons folder in case it was removed after SDL start

DrachenkoAnastasiia, 2016-11-07:
**Occurrence:**
Always

**Preconditions:**
SDL4.0 feature is enabled in .ini file, ' Icons' values is set as AppIconsFolder value, SDL and HMI are started. Icons folder is created.

**Steps to reproduce:**
1. Remove Icons folder
2. Register app via 4th protocol, app is set icon.

**Actual result:**
SDL does not create Icons folder.

**Expected result:**
SDL creates Icons folder to save icon for registered app

**Environment:**
SDL:
Repository: https://github.com/smartdevicelink/sdl_core/
Tag: 4.2.0
commit: a6c32b45832653d6b851be9b19355fa35f377e07
ATF version: 2.2 / 962e1eff8664d97a 

**Attachments:**

vvvakulenko, 2017-03-20:
**Reproducible**
SDL: HEAD of https://github.com/smartdevicelink/sdl_core/tree/feature/extended_policy_functionality 
(commit d5f63b4f66d9ab945d452dfb1c07bf7ee3ca781e)
ATF: HEAD of https://github.com/smartdevicelink/sdl_atf/tree/develop
(commit aa5beae6ab1726a9b43ed2fa15f66c20d7f7883f)

LitvinenkoIra, 2017-12-01:
Priority Low.
Loss of functionality in specific case.

iCollin, 2020-05-11:
Closed via #3359

PR #3359 (2020-05-11): SDL does not create icons folder in case it was removed (#2649)
REUPLOAD #2649 (branch deleted)

Fixes #1003

The check for existence directory of app icons folder
has been added to SetAppIconRequest.

If app icons folder does not exist, core will attempt to create it.

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
closed 2020-05-11 by iCollin