smartdevicelink/sdl_core #1004: App is disconnected due to PROTOCOL_VIOLATION when start audio streaming after rejected 2 times then accepted

KhrystynaDubovyk, 2016-11-07:
**Occurrence:**
Always

**Precondition:**
1.SDL is started
2.HMI is started

**Steps:**
1. Register Navi App
2. Activate App
3. Press "Audio Service - Show" button
4. Press "Start Service"
5. Press "Cancel" on popup
6. Press "Cancel" on popup
7. Press "Ok" popup
8. Press "Start File Streaming"

**Expected result:**
Audio Streaming is started.
Actual result:
App is unregistered with reason "PROTOCOL_VIOLATION".

**Environment:**
SDL repo: https://github.com/smartdevicelink/sdl_core.git
SDL branch: master
SDL commit: a6c32b4
HMI: https://github.com/LuxoftSDL/sdl_hmi: 01cb5b49883ed6288e34fcf7db24fd0439d08312
OS: Ubuntu 14.04 LTS

**Attachments:**

jacobkeeler, 2017-11-07:
@KhrystynaDubovyk @LuxoftAKutsan Looks like this issue still exists, but it may be related to the HMI.

When a `StartService(AUDIO)` is send to Core, it begins sending StartAudioStream once every second until a response is received from the HMI. This makes the HMI open several copies of the same "Would you like to start Audio Stream?" pop up on top of each other. If the user responds with `Cancel` to the first pop up, then responds with `Ok` of the other pop ups, then Core will receive a `failed` response followed by a `success` response. This is what seems to cause the `PROTOCOL_VIOLATION` message. Core should likely not be receiving several `StartAudioStream` responses.

Considering I haven't found any requirement/diagram where Core should be sending multiple `StartAudioStream` messages, I'm inclined to think this is an issue with Core, despite the fact that this behavior seems intentional.

dboltovskyi, 2017-11-09:
@jacobkeeler Please find diagram:
[image]

LuxoftAKutsan, 2017-11-13:
@jacobkeeler issue is not reproduced in develop in the same sequence as it described ( implicitly partially fixed). 
But main root cause is still exist in SDL and will be fixed.

jacobkeeler, 2017-11-22:
Closed with the merge of #1855

PR #1855 (2017-11-22): Fix application start streaming retry sequence
There was a problem with retry sequence which is triggered when user rejecets prompt regarding starting audio streaming on HMI. The problem is when user clicks Cancel then SDL sends new request in the sequence and not removes old one from queue. As a result we have two identical requests and each of them could trigger next iteration of retry sequence at the same time. As a result, next try could be started earlier than expected.

For example if streaming retry sequence has 3 retries with 1 second timeout:
- MOB -> SDL Request to start audio stream [... cut]
closed 2017-11-22 by jacobkeeler