smartdevicelink/sdl_core #1006: SDL responses with GENERIC_ERROR instead of UNSUPPORTED_RESOURCE

KhrystynaDubovyk, 2016-11-07:
**Preconditions:**
SDL is started

**Actions:**
Send any single UI-related RPC , UI interface is not supported by the system (script.lua from attachment could be used to reproduce)
SDL receives UI.IsReady (available=false) from HMI

**Actual result:**
SDL responds with GENERIC_ERROR, success=false (logs are in attachment)

**Expected result:**
SDL must respond "UNSUPPORTED_RESOURCE, success=false, info: UI is not supported by system" to mobile app

**Environment:**
Tag: 4.2.0
commit: a6c32b45832653d6b851be9b19355fa35f377e07
ATF version: 2.2 / 962e1eff8664d97a 

**Attachments:**

VVeremjova, 2017-03-17:
Attached script sends request not only on UI, but on TTS interface too. Attached updated script
[fixed_script.lua.tar.gz](https://github.com/smartdevicelink/sdl_core/files/850040/fixed_script.lua.tar.gz)

JackLivio, 2017-11-07:
When testing this issue I always receive `UNSUPPORTED_RESOURCE`, but depending on the RPC I either receive differing `SUCCCESS` results.

For example,

Show request response from core
```
{
   "info" : "Destination controller not found!",
   "resultCode" : "UNSUPPORTED_RESOURCE",
   "success" : true
}

```

Alert request response from core
```
{
   "info" : "Destination controller not found!",
   "resultCode" : "UNSUPPORTED_RESOURCE",
   "success" : false
}

```

AddCommand response from core
```
 {
   "info" : "Destination controller not found!",
   "resultCode" : "UNSUPPORTED_RESOURCE",
   "success" : true
}

```

ButtonSubscription Response from core (PRESET_0)
```
{
   "resultCode" : "UNSUPPORTED_RESOURCE",
   "success" : false
}

```

LuxoftAKutsan, 2017-11-20:
@JackLivio looks like this issue was indirectly fixed with 4028784

JackLivio, 2017-11-20:
After retesting with isReady:false instead of disabling the ui component on the hmi loading screen, I am able to see that this issue is non reproducible and the correct success cases are sent.

closed 2017-11-20 by JackLivio