smartdevicelink/sdl_core #1007: [Policies] Ford-specific keys are present in Genivi Policy DB - DeviceData

DrachenkoAnastasiia, 2016-11-07:
**Occurrence:**
Always

**Preconditions:**
1. Policy DB is absent on FS.

**Steps to reproduce:**
1. Start SDL

**Actual result:**
SDL creates DB, table "device" contains Ford-specific keys like "hardware", "firmware_rev" etc.

**Expected result:**
"DeviceData" secition is Ford-specific

**Note:** device_consent_group and consent_group also are present in the Genivi Policy DB

**Environment:**
SDL repo: https://github.com/smartdevicelink/sdl_core.git
SDL branch: master
SDL commit: a6c32b4
HMI: https://github.com/LuxoftSDL/sdl_hmi: 01cb5b49883ed6288e34fcf7db24fd0439d08312
OS: Ubuntu 14.04 LTS

**Attachments:**

jacobkeeler, 2017-11-07:
@DrachenkoAnastasiia @LuxoftAKutsan The issue that this describes still occurs, but I'm not certain that it was ever incorrect behavior in the first place.

dboltovskyi, 2017-11-09:
@jacobkeeler 
The defect was raised a quite a long time ago - in Dec-2015.
And at that date SDL had own PTU flow, but anyway Policy database contained redundand Ford-specific fields.
However in Feb-2017 SDL was extended with new EXTERNAL_PROPRIETARY flow functionality.
And this supposes presence of Ford-specific keys in Policy database.

So for now this issue is not actual.

LuxoftAKutsan, 2017-11-16:
```SDL creates DB, table "device" contains Ford-specific keys like "hardware", "firmware_rev" etc.```
When Extended policy was not in open source , keys "hardware", "firmware_rev" etc...  shouldn't be there.
But extended policy functionality is actually adding this keys.
Thats why this defect is not actual after merging Extended policy functionality

closed 2017-11-16 by jacobkeeler