smartdevicelink/sdl_core #1012: SDL does not respond info message in case GENERIC_ERROR watchdog timeout from HM

KhrystynaDubovyk, 2016-11-07:
**Occurrence:** 
Always

**Preconditions:**
1. Move IsReady_Template folder into sdl_atf\user_modules

**Steps to reproduce:**
Or: (Go to atf folder -> Run command: ./start.sh ../ATF_VehicleInfo_IsReady_missing_SingleRPC_empty.lua)

1. RegisterApp in case VehicleInfo.IsReady=false
2. Activate app
3. Send READDID with "didLocation":[56832],"ecuName":2000 from mobile app
4. HMI does not respond during SDL`s watchdog

**Actual result:**
SDL respond
{"resultCode":"GENERIC_ERROR","success":false}

to mobile app.

**Expected result:**
SDL should respond
{"resultCode":"GENERIC_ERROR","success":false,"info":"VehicleInfo component does not respond"}

**Notes:** It also happens with other APIs: , GetDTCs, DiagnosticMessage, SubscribeVehicleData, GetVehicleData, UnsubscribeVehicleData

**Environment:**
SDL branch: master
SDL commit: a6c32b4
HMI: HTML5
OS: Ubuntu 14.04 LTS
Transport: WiFi
Mobile device: HTC One
Mobile OS: Android 4.3
Mobile App version: SPT v.2.1.81 Code: 239
Mobile App type (media/non-media/navi/any): any

**Attachments:**

jacobkeeler, 2017-11-02:
@KhrystynaDubovyk The ATF script that was supposed to be included with this issue was never uploaded.

jacobkeeler, 2017-11-16:
Fixed with the merge of #1640

closed 2017-11-16 by jacobkeeler