smartdevicelink/sdl_core #1018: [SDL4.0] SDL sends wrong parameter names in OnSystemRequest(query_apps)

OVikhrov, 2016-11-07:
**Occurrence**: always

Preconditions: SDL4.0 feature is enabled in .ini file, SDL and HMI are started.

**Steps**:
1. Register app via 4th protocol.

**Expected result:** 
After receiving OnHMIStatus("FULL") from mobile , SDL sends OnSystemRequest(QUERY_APPS) with header
_{
	"HTTPRequest": {
		"headers": {
			"ContentType": "application/json",
			"ConnectTimeout": <int value from policy table:timeout_after_x_seconds>,
			"DoOutput": true,
			"DoInput": true,
			"UseCaches": false,
			"RequestMethod": "*GET*",
			"ReadTimeout": <int value from policy table:timeout_after_x_seconds>,
			"InstanceFollowRedirects": false,
			"charset": "utf-8",
			"Content_Length": <value of the length of the "body" string>
		}
		_<*no* "body" parameter>_
	}
}_

**Actual result:** 
SDL sends "ConnectTimout" name instead of "ConnectTimeout", "InstanceFollowRedirect" instead of "InstanceFollowRedirects", "Content_Lenght" instead of "Content_Length".

**Notes:** 
To reproduce defect using ATF extract files_for_scrip_execution_OnSystemRequest_query_apps.tar.gz in ATF_build folder and execute script OnSystemReques_query_apps.lua.

**Attachments:**

vvvakulenko, 2017-03-20:
**Reproducible**
SDL: HEAD of https://github.com/smartdevicelink/sdl_core/tree/feature/extended_policy_functionality
(commit d5f63b4f66d9ab945d452dfb1c07bf7ee3ca781e)
ATF: HEAD of https://github.com/smartdevicelink/sdl_atf/tree/develop
(commit aa5beae6ab1726a9b43ed2fa15f66c20d7f7883f)

LitvinenkoIra, 2017-12-01:
Priority Low.
Not affect feature functionality.

iCollin, 2021-03-31:
Closing as no longer reproducible, it seems these parameters are now spelled correctly.

Please let me know if you believe this issue is still actual.

closed 2021-03-31 by iCollin