smartdevicelink/sdl_core #1026: [Policies] SDL doesn't send "SDL.OnAppPermissionChanged{appID} to HMI

Itileda, 2016-11-07:
**Occurrence:** 
Always

**Preconditions:** 
SDL is built with Flag ENABLE_EXTENDED_POLICY = ON. SDL and HMI are started.
SPT is configured to send https://adc.luxoft.com/svn/APPLINK/doc/technical/testing/Policies/PTU_for_manual_test_cases/PTFromCloud_smoke%20test.json as reply to SDL's OnsyustemRequest. To do it
1. place https://adc.luxoft.com/svn/APPLINK/doc/technical/testing/Policies/PTU_for_manual_test_cases/PTFromCloud_smoke%20test.json to sdcard
2. Start SPT (SDL is not started), click AddSession
3. Goto Menu > Set up Policy files
4. in case of Ubuntu SDL is under testing, check the checkbox "Override Policy Table Update data" and provide path to uploaded json on sdcard.
in case of HeadUnit SDL is under testing, uncheck allcheckboxes

**Steps:** 
Register App:
appName: SyncProxyTester Null
appID: 584421907

**Expected result:** 
PTU is performed. Right after receiving SystemRequest, SDL should apply update to DB, notify App about changed RPCs with OnAppPermissionsChange.

**Actual result:** 
SDL does not send to HMI OnAppPermissionsChanged notification

**Environment** 
Branch: Repository: https://github.com/smartdevicelink/sdl_core/ Tag: 4.2.0
HMI: 3acad9e9673fb479a
SDL: a6c32b45832653d6b851be9b19355fa35f377e07
ATF version: N/A
OS: Ubuntu 14.04 TTS x64
Transport: USB
Mobile device: Acer B1-710
Mobile OS: Android 4.1.2
Mobile App version: SPT ver 2.1.84 code 242
Mobile App type: Navi
Virtual machine using: N/A

JackLivio, 2017-04-18:
@mghiumiusliu If this issue is scoped to be fixed as part of an upcoming release could you please add the appropriate milestone?

jacobkeeler, 2017-11-03:
@LuxoftAKutsan @mghiumiusliu This issue still appears to be present on develop, I do not see any `OnAppPermissionsChanged` notification on the HMI side after performing a PTU using the provided `PTFromCloud_smoke test.json`. 

Note: I'm using `EXTENDED_POLICY=PROPRIETARY`, since `ENABLE_EXTENDED_POLICY` is obsolete.

JackLivio, 2017-11-03:
Verified that OnAppPermissionsChanged was sent after revoking app permissions via policy table update.

closed 2017-11-03 by JackLivio