smartdevicelink/sdl_core #1027: [Policy]: SDL does not write UserFriendlyMessages to DB

OVikhrov, 2016-11-07:
Occurrence: 
Always

Preconditions: 
1. SDL built with DEXTENDED_POLICY: ON
2. First SDL life cycle.

Steps to reproduce:
1. Start SDL.
2. Register app and perform PTU.

Actual result:
During PTU HMI sends GetUserFriendlyMessage to SDL but SDL responds "messages":[
{"messageCode":"StatusPending"}
] where "messages" is re-sent part of request. UserFriendlyMessage are absent in DB.

Expected result:
SDL should write UserFriendlyMessage to DB and correctly respond to HMI GetUserFriendlyMessage request.

Environment:
SDL: https://github.com/smartdevicelink/sdl_core.git: a6c32b45832653d6b851be9b19355fa35f377e07
ATF: https://github.com/smartdevicelink/sdl_atf.git: 962e1eff8664d97af5e59ec4e22f39b9ea76f5eb
OS: Ubuntu 14.04 LTS

Attachments:

JackLivio, 2017-04-18:
@mghiumiusliu If this issue is scoped to be fixed as part of an upcoming release could you please add the appropriate milestone?

jacobkeeler, 2017-09-07:
Fixed with the merge of #1615

PR #1615 (2017-09-07): Fix Consumer Friendly Messages removing from PT
In case PTU comes with omitted `consumer_friendly_messages` param, SDL should maintain current `consumer_friendly_messages` section in Local PT. 
Currently SDL removes all `consumer_friendly_messages` from PT even if this param was omitted in PTU.

Related PR in another branch: #1415 
Fixes #1027
closed 2017-09-07 by jacobkeeler