smartdevicelink/sdl_core #1028: [Security]: App continue unprotected stream if start service as protected during active streaming

Itileda, 2016-11-07:
**Occurrence:** 
Always

**Preconditions:** 
1. Core, HMI started.
2. Navi app registered and activated on HMI.

**Steps to reproduce:**
1. Strart unprotected Video service (accept with pop up on HMI).
2. Start video/audio streaming
3. Start protected video/audio service.

**Actual result:**
SDL continue stream as unprotected (see logs).

**Expected result:**
The streamed data should become protected.

**Environment**
Branch: Repository: https://github.com/smartdevicelink/sdl_core/ Tag: 4.2.0
HMI: 3acad9e9673fb479a
SDL: a6c32b45832653d6b851be9b19355fa35f377e07
ATF version: N/A
OS: Ubuntu 14.04 TTS x64
Transport: USB
Mobile device: Acer B1-710
Mobile OS: Android 4.1.2
Mobile App version: SPT ver 2.1.84 code 242
Mobile App type: Navi
Virtual machine using: N/A

JackLivio, 2017-04-18:
@mghiumiusliu If this issue is related specifically to a release being worked on, can you please add the appropriate milestone?

LitvinenkoIra, 2017-12-01:
Priority High.
Significant loss of functionality on SDL side with standard use-case

AKalinich-Luxoft, 2017-12-08:
After deep investigation of that issue there was found out that it is not related to SDL.
SDL correctly starts secured video service and sends ACK with protection flag ON to mobile however mobile continues sending of unprotected data. SDL just transfers this data to HMI as is. Problem should be fixed on mobile side.

**Analysis**
Defect was reproduced using provided steps. In attached [Logs.tar.gz](https://github.com/smartdevicelink/sdl_core/files/1542550/Logs.tar.gz) archive there are 2 files:

1. SDL log of described test case
Line 21932 - Starting of unprotected video service
Line 45589 - Starting of protected video service
Line 45628 - Starting of handshake
Line 47699 - Last frame which should be unencrypted - OK
Line 47773 - Handshake finished successfully
Line 47796 - Set session protection flag to ON
Line 47897 - First frame which should be encrypted - still unencrypted (PROTECTION: OFF) - NOT OK

2. Wireshark log with captured data from described test case
10.42.0.1 - SDL; 10.42.0.44 - Mobile
Before starting of protected service size of video streaming data packages is 1078 bytes
Packet with no 1087 - sending StartSessionACK with encrypted:true
After this packet all next packages from Mobile (Source = 10.42.0.44) should have different size and encrypted content, however they are still the same - NOT [OK]

dboltovskyi, 2017-12-20:
ATF test script to check if SDL Core works correctly will be created once [ATF Security feature](https://github.com/smartdevicelink/sdl_atf/tree/feature/security) is implemented in merged into develop branch.

mrapitis, 2017-12-21:
Will be tested when ATF security feature has been implemented as a dependency exists.

dboltovskyi, 2018-08-22:
Mentioned [ATF Security](https://github.com/smartdevicelink/sdl_atf/tree/feature/security) feature was successfully implemented. 
It allows setting up secure connection for RPC or Audio/Video services.
However currently in ATF there is no functionality to check streaming data.
Additional proposal ([0125-atf-videostreaming-full-support](https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0125-atf-videostreaming-full-support.md)) describes this.
It's required in order to verify current issue automatically using ATF.

iCollin, 2021-07-09:
Closing as not reproducible.

In testing I found similar results to Andrii's but the second session started as protected has `PROTECTION: ON`.

closed 2021-07-09 by iCollin