smartdevicelink/sdl_core #1029: [Memory leaks]: SDL does not release memory after sending AddCommand limit exhausted

Itileda, 2016-11-07:
**Occurrence:** 
Always

**Preconditions:** 
1. Core, HMI started.

**Steps to reproduce:**
1. Register and activate app.
2. Execute AddCommand_limitExhausted.xml (attached).

**Actual result:**
After finishing test SDL does not release memory (see screenshots)

**Expected result:**
SDL should release memory.

**Environment**
Branch: Repository: https://github.com/smartdevicelink/sdl_core/ Tag: 4.2.0
HMI: 3acad9e9673fb479a
SDL: a6c32b45832653d6b851be9b19355fa35f377e07
ATF version: N/A
OS: Ubuntu 14.04 TTS x64
Transport: USB
Mobile device: Acer B1-710
Mobile OS: Android 4.1.2
Mobile App version: SPT ver 2.1.84 code 242
Mobile App type: Navi
Virtual machine using: N/A

JackLivio, 2017-04-18:
@mghiumiusliu I see this one doesn't have a project associated with it so I'm not sure if there is a milestone related to this fix, or if it is just a general issue. Please place milestone if it is related to a release.

mghiumiusliu, 2017-11-01:
@JackLivio This is part of stabilization fixes we provided in June.

jacobkeeler, 2017-11-02:
Closing with the merge of #1626

closed 2017-11-02 by jacobkeeler