smartdevicelink/sdl_core #1030: [API] SDL sends OnAppInterfaceUnregistered(DRIVER_DISTRACTION_VIOLATION) to app when receives OnExitApplication(DRIVER_DISTRACTION_VIOLATION) from HMI

OVikhrov, 2016-11-07:
**Occurrence:** 
Always

**Precondition:**
SDL Core and HMI are started. App is registered
App is None

**Steps:**
From HMI: send BasicCommunication.OnExitApplication", {reason = "DRIVER_DISTRACTION_VIOLATION", AppId=ID of app in the precondition}
Note: You can use attached lua file to reproduce this defect automatically.

**Expected result:**
SDL doesn't send OnAppInterfaceUnregistered(reason = "DRIVER_DISTRACTION_VIOLATION") to mobile app
App is not registered

**Actual result:**
SDL sends OnAppInterfaceUnregistered(reason = "DRIVER_DISTRACTION_VIOLATION") to mobile app
App is not registered

**Environment:**
Release: 4.2.0 (master marge)
Repository: https://github.com/smartdevicelink/sdl_core
Commit: a6c32b45832653d6b851be9b19355fa35f377e07
OS: Ubuntu 14.04

**Attachments:**

JackLivio, 2017-04-18:
@mghiumiusliu If this issue is related to the 4.4.1 milestone can you please add the milestone label to this issue?

LitvinenkoIra, 2017-12-01:
Priority Low.
Sending redundant notification. Not affect feature functionality.

HSavynetska, 2019-11-19:
The defect is not reproduced in the current develop

Environment:
Script: https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/2112
SDL build: https://github.com/smartdevicelink/sdl_core/tree/develop ( 891497b  )
ATF: develop branch

iCollin, 2021-03-31:
Closed as not reproducible against the first 7.1 release candidate.

Please let me know if you believe this issue is still actual.

closed 2021-03-31 by iCollin