smartdevicelink/sdl_core #1031: SDL transfer OnKeyboardInput notification to not active App when there is no active PerformInteraction(KEYBOARD)

Itileda, 2016-11-07:
**Occurrence:** 
Always

**Preconditions:** 
1. Core, HMI started. 
2. App is registered and deactivated on HMI (has LIMITED level)
3. OnKeyboardInput notification is allowed to the App from LIMITED
4. Choise set with id 1 is created.

**Steps to reproduce:**
1. Send PerformInteraction(ICON_ONLY)
2. During processing request send OnKeyboardInput notification

**Actual result:**
SDL resends OnKeyboardInput() to the App

**Expected result:**
In case there is no active PerformInteraction(KEYBOARD), SDL should resend OnKeyboardInput only to App that is currently in FULL.
Note: SDL also transfer OnKeyboardInput notification to the BACKGROUNDED app

**Environment**
SDL repo: https://github.com/smartdevicelink/sdl_core/
SDL branch: a6c32b45832653d6b851be9b19355fa35f377e07
HMI: HTML5
ATF version: 2.2
ATF repo: https://github.com/smartdevicelink/sdl_atf/
ATF branch: 962e1eff8664d97af5e59ec4e22f39b9ea76f5eb 
Scripts repo: https://github.com/CustomSDL/sdl_atf_test_scripts/tree/OS_DRAFT
Scripts branch: 18cce42
OS: Ubuntu 14.04.01
Transport: Wi-Fi
Mobile device: -
Mobile OS: -
Mobile App version: -
Mobile App type (media/non-media/navi/any): any
Virtual Machine using (yes/no): no

JackLivio, 2017-04-18:
@mghiumiusliu If this has an associated milestone can you please attach to this issue?

LitvinenkoIra, 2017-12-01:
Priority is hight: significant loss of functionality on SDL side with standard use-case

LitvinenkoIra, 2017-12-05:
Already fixed in develop c656bc51132b66f086cf56e590d63431756aa524

BSolonenko, 2018-05-10:
Works as expected on develop https://github.com/smartdevicelink/sdl_core/commit/df9b67026d90f0f9cdf04bec7dd3c7666a25db87

yang1070, 2018-05-15:
what about the cases for layout of ICON_WITH_SEARCH and LIST_WITH_SEARCH which also allow KEYBOARD to be shown when user touches the search? shall the notification be sent in these two layout and app is not in full?

AStasiuk, 2018-05-25:
@BSolonenko could you please create manual test case as @yang1070  requested.

KhrystynaDubovyk, 2018-05-29:
hello, @yang1070 
regarding the question: 

> what about the cases for layout of ICON_WITH_SEARCH and LIST_WITH_SEARCH which also allow KEYBOARD to be shown when user touches the search? shall the notification be sent in these two layout and app is not in full?

it looks like currently we do not have any requirements with expected behaviour for mentioned cases. 
it could be investigated rosen as a separate issue.

BSolonenko, 2018-05-29:
hello, @yang1070 
regarding the question:

> Also how can manually I reproduce the issue? Follow the steps described in the thread, if I send PerformInteraction(ICON_ONLY), how can I trigger OnKeyboardInput from HMI? There shall be no keyboard for icon_only and list_only layout.

**Preconditions:**
1. Core, HMI started.
2. OnKeyboardInput notification is allowed by policy for the App in LIMITED HMI level
3. App is registered, activated and deactivated on HMI (has LIMITED level)
4. Choise set with id 1 was created via CreateInteractionChoceSet

**Steps to reproduce:**
1. Send PerformInteraction(ICON_ONLY, ID = 1)
2. During PI processing, toggle on-screen keyboard on HMI, enter some text and click Search to send OnKeyboardInput notification
3. Check in HMI console (press F12) that OnKeyboardInput was sent to SDL
4. Check that app in LIMITED is not received OnKeyboardInput notification

GetmanetsIrina, 2018-07-30:
Script for reproducing https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Defects/5_0/1031_SDL_transfer_OnKeyboardInput_notification_to_not_active_App_when_there_is_no_active_PerformInteraction_KEYBOARD.lua

Ypostolov, 2019-11-29:
The defect is not reproduced on the current develop

Environment:
SDL build: https://github.com/smartdevicelink/sdl_core/tree/develop (891497b )
ATF: develop branch
Script to reproduce defect: [https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Defects/5_0/1031_SDL_transfer_OnKeyboardInput_notification_to_not_active_App_when_there_is_no_active_PerformInteraction_KEYBOARD.lua]

Test script created on the develop branch #6f43bfd

jacobkeeler, 2020-04-09:
Closing due to https://github.com/SmartDeviceLink/sdl_core/issues/1031#issuecomment-559776980

closed 2020-04-09 by jacobkeeler