smartdevicelink/sdl_core #1032: CreateInteractionChoiceSet: core successfully creates choice set with duplicate vrCommands/menuName inside it

OVikhrov, 2016-11-07:
**Occurrence:** 
Always

**Preconditions:** 
1. Core, HMI started. 
2. Application is registered, HMI level = FULL.

**Step to Reproduce:**
1. Send CreateInteractionChoiceSet with duplicate vrCommands, other parameters are valid.

**Expected result:** 
Choice set isn't created and SDL response resultCode = DUPLICATE_NAME, success=false to mobile.

**Actual result:** 
SDL sends two VR.AddCommand with the same vrCommands(Choice1059 in attached log).

**Environment:**
SDL repo: https://github.com/smartdevicelink/sdl_core/
SDL branch: a6c32b45832653d6b851be9b19355fa35f377e07
HMI: HTML5
OS: Ubuntu 14.04.01
Transport: Wi-Fi
Mobile device: Lenovo P1ma40
Mobile OS: Android V5.1
Mobile App version: Luxoft SPT
Mobile App type (media/non-media/navi/any): media + non-media

**Attachments:**

JackLivio, 2017-04-18:
@mghiumiusliu If this has an associated milestone could you please attach to this issue?

LitvinenkoIra, 2017-12-01:
Priority is Medium: Wrong result code received from SDL

HSavynetska, 2018-06-12:
Defect is not reproduce on current develop(91123c3cae83ad41e1f5f37d37e46e9f07264d1b)
Verified with script(Script under review ): https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/1914

closed 2018-06-15 by HSavynetska

GetmanetsIrina, 2018-07-30:
Script for reproducing https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Defects/5_0/1032_CreateInteractionChoiceSet_core_successfully_creates_choice_set_with_duplicate_vrCommands_menuName_inside_it.lua