smartdevicelink/sdl_core #1033: SDL forwards OnButtonPress notification of CUSTOM_BUTTON to BACKGROUND App.

Itileda, 2016-11-07:
**Occurrence:** 
Always

**Preconditions:** 
1. Core, HMI started. 
2. App is registered on HM and has HMIlevel BACKGROUND

**Steps to reproduce:**
1. Send from HMI OnButtonEvent(CUSTOM_BUTTON, SHORT/LONG)

**Actual result:**
SDL resend it to BACKGROUND App.

**Expected result:**
SDL should resend OnButtonEvent notification to either FULL non-media App or FULL/LIMITED media App (see https://adc.luxoft.com/contour/perspective.req?projectId=65&docId=209491)

**Note:** to reproduce can be used lua script - attached.
for run:
create folder user_modules in <ATF-2.2> folder.
copy content of https://adc.luxoft.com/svn/APPLINK/doc/technical/testing/TestCases/lua/user_modules/ to user_modules
run ./start.sh ATF_OnButtonEvent.lua

**Environment**
Repository: https://github.com/smartdevicelink/sdl_core/
Tag: 4.2.0
commit: a6c32b45832653d6b851be9b19355fa35f377e07
ATF version: 2.2 / 962e1eff8664d97a
NOTE: In order for the script to run please change line
from <<Test = require('user_modules/OnButtonPress_connecttest') >>
to <<Test = require('connecttest')>>

JackLivio, 2017-04-18:
@mghiumiusliu Can you please add the appropriate milestone? Not sure if it is still 4.4.1

LitvinenkoIra, 2017-12-01:
Priority is hight: significant loss of functionality on SDL side with standard use-case

LitvinenkoIra, 2017-12-05:
Already fixed in develop 7836d78b7f28ec7c54bef420f90449583a4e18d5

jacobkeeler, 2018-03-02:
Fixed with the merge of https://github.com/smartdevicelink/sdl_core/issues/1111

closed 2018-03-02 by jacobkeeler