smartdevicelink/sdl_core #1035: SDL doesn't send "REJECTED" code to mobile app when activating app from HMI with activate Carplay/GAL.GE

OVikhrov, 2016-11-07:
**Occurrence:** 
Always

**Precondition:** 
1.SDL is started
2.HMI is started
3.App is registered via Wifi.
4. App is in "FULL" HMI Level.

**Steps:**
1. Click Exit Application -> Send "OnDeactivate notification send = true" from HMI
2. After that send SDL.ActivateApp from HMI
3. Observer the result

**Expected result:**
SDL should send "REJECTED" code to HMI.

**Actual result:**
SDL sends "SUCCESS" code to HMI.

**Environment:**
Branch: Repository: https://github.com/smartdevicelink/sdl_core/ Tag: 4.2.0
HMI: 3acad9e9673fb479a
SDL: a6c32b45832653d6b851be9b19355fa35f377e07
ATF version: N/A
OS: Ubuntu 14.04 TTS x64
Transport: USB
Mobile device: Acer B1-710
Mobile OS: Android 4.1.2
Mobile App version: SPT ver 2.1.84 code 242
Mobile App type: Navi

**Attachments:**

JackLivio, 2017-04-18:
@mghiumiusliu Is this for milestone 4.4.1? If so could you please attach that milestone to this issue?

LuxoftAKutsan, 2017-12-01:
Priority is hight
significant loss of functionality on SDL side with standard use-case

BSolonenko, 2018-05-10:
Works as expected on develop https://github.com/smartdevicelink/sdl_core/commit/df9b67026d90f0f9cdf04bec7dd3c7666a25db87

closed 2018-05-15 by Itileda

GetmanetsIrina, 2018-07-30:
Script for reproducing https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Defects/5_0/1035_SDL_doesnt_send_REJECTED_code_to_mobile_app_when_activating_app_from_HMI_with_activate.lua