smartdevicelink/sdl_core #955: Incorrect response on send SystemRequest (file name - /test)

DrachenkoAnastasiia, 2016-11-07:
**Occurrence:**
Always
**Preconditions:**
1. SDL and HMI are started
2. App is registered
3. App HMIlevel=FULL
**Steps to reproduce:**
1. Send SystemRequest (choose any file from sdcard on mobile), specify request type - PROPRIETARY, Use file name - /test 
**Expected result:**
SDL responses with INVALID DATA result code
**Actual result:**
SDL responses with REJECTED result code

**Environment:**
Verified on release SDL_RB_B3.37.
Failed
SDL repo: https://github.com/smartdevicelink/sdl_core/
SDL branch: a6c32b45832653d6b851be9b19355fa35f377e07
HMI: HTML5
ATF version: -
ATF repo: -
ATF branch: -
Scripts repo: -
Scripts branch: -
OS: Ubuntu 14.04.01
Transport: Wi-Fi
Mobile device: Lenovo P1ma40
Mobile OS: Android V5.1
Mobile App version: Luxoft SPT
Mobile App type (media/non-media/navi/any): any
Virtual Machine using (yes/no): no

**Attachments:**

LitvinenkoIra, 2017-12-01:
Priority Low.
Specific use-case related to used file name

robinmk, 2018-06-21:
@LitvinenkoIra , why should the response be INVALID_DATA? 
Please add more details to the issue .

KhrystynaDubovyk, 2018-07-31:
@robinmk, I guess this behaviour (and expected result) is based on the following requirements:

1. SDL must respond with 'INVALID_DATA, success:false' resultCode 
in case the name of file that the app requests to upload on the system 
contains `"/"` symbol (example: fileName: "../123.jpg")
2. SDL must respond with 'INVALID_DATA, success:false' resultCode 
in case the name of file that the app requests to upload on the system contains `"\"` (backslash) symbol (example: fileName: "..\123.jpg"

**Impacted RPC:**
-> PutFile (`<syncFileName>`)
-> SystemRequest (`<fileName>`)

robinmk, 2018-07-31:
@KhrystynaDubovyk , Thank you for clarifying. Now I understand the problem.
Since the response is `REJECTED`, I agree that its a low priority bug.

E-SAITO-TMC, 2020-04-23:
@theresalech -san This issue is a medium priority for Toyota. We tried it about 50 times and couldn't reproduce it. I would like to know if this issue occur in the latest version. And could you please include this issue in your plan for Q4 Core ?  Thank you.

theresalech, 2020-04-23:
Hi @E-SAITO-TMC, we have documented that this is medium priority for Toyota, and will include for Steering Committee review and vote when planning the contents for the Q4 Core release, if issue is still reproducible.

@DrachenkoAnastasiia, can you please advise if you are seeing this issue when using Core 6.1?

ychernysheva, 2020-04-27:
@theresalech This issue is not actual for Core 6.1. There is a need in actualization of corresponding test script, after that it can be delivered on Livio Team review.

theresalech, 2020-05-11:
Hi @ychernysheva, please let us know once the corresponding test script is ready for review.  We can then review the test script and close this Core issue as not reproducible.

dboltovskyi, 2020-05-11:
@theresalech Please notice PR with script is ready for PM review: https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/2071

ShobhitAd, 2020-06-12:
Issue not reproducible in latest develop(https://github.com/smartdevicelink/sdl_core/commit/ef2b88520b4224390ffd332f737fdf5573def60e).

closed 2020-06-12 by ShobhitAd