smartdevicelink/sdl_core #956: putFile does not copy file with .\\ before the name to AppStorageFolder

DrachenkoAnastasiia, 2016-11-07:
**Occurrence:**
Always
**Preconditions:**
1. SDL and HMI are started
2. App is registered("Set app icon" after registration checkbox is unchecked)
3. App HMIlevel=FULL
**Steps to reproduce:**
1. Send PutFile (choose any file from sdcard on mobile), specify Sync File Name - .
syncFileName (dot two backslashes before the name) 
**Actual result:**
the SPT displays PutFile(response):SUCESS, but chosen file didn't appear in AppStorageFolder
**Problem can be reproduced with ATF script:**

**Environment:**
Verified on release SDL_RB_B3.37.
Failed
SDL repo: https://github.com/smartdevicelink/sdl_core/
SDL branch: a6c32b45832653d6b851be9b19355fa35f377e07
HMI: HTML5
ATF version: -
ATF repo: -
ATF branch: -
Scripts repo: -
Scripts branch: -
OS: Ubuntu 14.04.01
Transport: Wi-Fi
Mobile device: Lenovo P1ma40
Mobile OS: Android V5.1
Mobile App version: Luxoft SPT
Mobile App type (media/non-media/navi/any): any
Virtual Machine using (yes/no): no

**Attachments:**

LitvinenkoIra, 2017-12-01:
Priority Low.
Specific use-case related to used file name

robinmk, 2018-06-21:
As long as the file is not copied (to prevent file traversal issue), agree that it is a lower priority bug. However, if the file is copied to another folder it should be fixed as a high priority item.

ValeriiMalkov, 2018-08-01:
### The bug isn't reproduced. 
### Note:warning: 
In Unix-like operating systems, any file or folder that starts with a dot character (for example, /home/user/.config), commonly called a dot file or dotfile, is to be treated as hidden – that is, the `ls` command does not display them unless the `-a` flag `ls -a` is used.

AStasiuk, 2018-08-02:
Closed as not reproducible.

closed 2018-08-02 by AStasiuk