smartdevicelink/sdl_core #958: [Policy]: SDL does not send RequestType:HTTP in OnSystemRequest to app

KhrystynaDubovyk, 2016-11-07:
**Occurrence:**
Always

**Preconditions:**
1. SDL built with DEXTENDED_POLICY: OFF
2. Core, HMI started.
3. First SDL life cycle.
4. Policy autoreplay checkbox is unchecked on app.

**Steps to reproduce:**
1. Register App

**Actual result:**
SDL send OnSystemRequest with requestType:PROPRIETARY and fileType:Json to app.

**Expected result:**
SDL should send OnSystemRequest with requestType:HTTP and fileType:BINARY to app.

**Enviroment**
SDL branch: a6c32b45832653d6b851be9b19355fa35f377e07
HMI: HTML5
ATF version: -
ATF repo: -
ATF branch: -
Scripts repo: -
Scripts branch: -
OS: Ubuntu 14.04.01
Transport: Wi-Fi
Mobile device: Lenovo P1ma40
Mobile OS: Android V5.1
Mobile App version: Luxoft SPT
Mobile App type (media/non-media/navi/any): any
Virtual Machine using (yes/no): no

LitvinenkoIra, 2017-12-01:
Priority Medium.
Loss of functionality on SDL side with standard use-case.

JackLivio, 2018-03-01:
Issue outdated. Run with HTTP policies and onSystemRequest sends HTTP and binary. Not sure where the issue was fixed but it no longer exists on master or develop.

closed 2018-03-01 by JackLivio