smartdevicelink/sdl_core #959: HMILevel is not resumed to LIMITED for non-media applications

KhrystynaDubovyk, 2016-11-07:
**Occurrence:**
Always

**Preconditions:**
1. Values configured in .ini file:
AppSavePersistentDataTimeout =10000;
ResumptionDelayBeforeIgn = 30;
ResumptionDelayAfterIgn = 30;
ApplicationResumingTimeout = 5000
2. Core and HMI are started.
3. Non-media application(COMMUNIATION) is registered and activated. -> HMI level = FULL
4. Go to menu Apps. HMI level of application becomes LIMITED
5. Stop WiFi connection.

**Steps:**
1. Press "Go To CD" -> HMI sends OnEventChanged(AUDIO_SOURCE, isActive: true) notification to SDL.
2. Activate WiFi connection.
3. Wait ApplicationResumingTimeout to expire.

**Expected result:**
1. SDL must resume HMILevel for non-media app.
Actual result:
1. HMI level becomes BACKGROUND, audioStreamingState : NOT_AUDIBLE

**Notes:**
1. If before ApplicationResumingTimeout is expired SDL receives OnEventChanged(AUDIO_SOURCE, isActive: false), SDL resumes HMI level.
2. The same situation is observed and at ign_off -> ign_on

**Enviroment**
SDL branch: a6c32b45832653d6b851be9b19355fa35f377e07
HMI: HTML5
ATF version: -
ATF repo: -
ATF branch: -
Scripts repo: -
Scripts branch: -
OS: Ubuntu 14.04.01
Transport: Wi-Fi
Mobile device: Lenovo P1ma40
Mobile OS: Android V5.1
Mobile App version: Luxoft SPT
Mobile App type (media/non-media/navi/any): Non-media application(COMMUNIATION)
Virtual Machine using (yes/no): no

LitvinenkoIra, 2017-12-01:
Priority Medium.
Standard use-case related to resumption data of one application type.

GetmanetsIrina, 2018-05-21:
Script for reproducing https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Defects/4_6/959_resumption_limited_non_media.lua