smartdevicelink/sdl_core #960: SDL retry send StartStream/StartAudioStream less on one time than configured in .ini file

DrachenkoAnastasiia, 2016-11-07:
**Occurrence:**
Always
**Preconditions:**
1. Core, HMI started.
2. StartStreamRetry = 3, 3 in .ini file.
3. Navi app registered and activated on HMI.
**Steps to reproduce:**
1. Start Audio/Video service.
2. Press cancel on HTML5 HMI (emulate REJECT from HMI) => SDL start retry sequence.
3. Press Cancel on all appeared pop-ups. 
**Actual result:**
After unsuccessful StartStream/StartAudioStream (first REJECT) SDL retry to start 2 times instead 3 times.
**Expected result:**
The counting of retries should be started after first unsuccessfuly response.

**Environment:**
Reproduces
Release: 4.2.0 (master marge)
Repository: https://github.com/smartdevicelink/sdl_core
Commit: a6c32b45832653d6b851be9b19355fa35f377e07
OS: Ubuntu 14.04
Transport: WiFi
App type: navi
App: Luxoft SPT
Device: HTC One 
**Attachments:**

AStasiuk, 2018-06-12:
Reopened - because fix is ready and PR should be reviewed by project maintainer.

jacobkeeler, 2020-05-18:
Closing with the merge of #2553

PR #2553 (2020-05-18): Fix Start Stream retry count
Original PR: #2210 
Fixes #960 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Summary
Fix Start Stream retry count

### Changelog
##### Bug Fixes
* Fix Start Stream retry count

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
closed 2020-05-18 by jacobkeeler