smartdevicelink/sdl_core #961: [Policy]: SDL dos not select url from PT for specified appID during GetURLs request.

KhrystynaDubovyk, 2016-11-07:
**Occurrence:**
Always

**Preconditions:**
1. SDL built with DEXTENDED_POLICY: ON
2. Core, HMI started using attached sdl_preloaded_pt.json.
3. SPT is configured to send attached PTU - PTFromCloud_smoke%20test__2.json.

**Steps to reproduce:**
1. Register App with appID "584421907" - PTU trigered successfully.
2. Register app "Base", appID = 12345.

**Actual result:**
After Registration second app SDL triger PTU. HMI sends GetURLS request to SDL to get url for second app. SDL responds: "url":"URL is not found". PTU is not finished successfull becouse OnSystemRequest is sent without url.

**Expected result:**
SDL should select url for current app from PT.

**Enviroment**
SDL branch: release/4.2.0
SDL commit: fa05b5f5b1af6fae7d684a5cc156ec2e68dbc197
HMI: HTML5
ATF version: commit 962e1e
OS: Ubuntu 14.04 LTS
App version: - 2.1.80
App type: - Alice SPT

theresalech, 2017-07-24:
Per Luxoft, this is addressed via PR #1539

JackLivio, 2017-08-01:
@KhrystynaDubovyk The content of this issue does not match the title or the pr associated with it.

JackLivio, 2017-08-01:
fix is merged

closed 2017-08-01 by JackLivio