smartdevicelink/sdl_core #963: SDL forwards OnButtonPress(CUSTOM_BUTTON) with wrong appID to current App.

DrachenkoAnastasiia, 2016-11-07:
**Occurrence:**
Always

**Preconditions:**
1. Core, HMI started.
2. App is registered on HM and has HMIlevel BACKGROUND

**Steps to reproduce:**
1. Send from HMI OnButtonPress(CUSTOM_BUTTON, SHORT/LONG, wrong appID)

**Actual result:**
SDL resends it to currently registered App.

**Expected result:**
SDL must ignore the notification. 

**Environment:**
Issue is still reproduced on:
SDL:
Repository: https://github.com/smartdevicelink/sdl_core/
Tag: 4.2.0
commit: a6c32b45832653d6b851be9b19355fa35f377e07
ATF version: 2.2 / 962e1eff8664d97a
NOTE: In order for the script to run please change line
from <<Test = require('user_modules/OnButtonPress_connecttest') >>
to <<Test = require('connecttest')>>

**Note:**
to reproduce can be used lua script - attached.
for run:
    create folder user_modules in <ATF-2.2> folder.
    copy content of https://adc.luxoft.com/svn/APPLINK/doc/technical/testing/TestCases/lua/user_modules/ to user_modules
    run ./start.sh ATF_OnButtonPress.lua

**Attachments:**

LitvinenkoIra, 2017-12-01:
Priority Medium.
Processing and resending notification in case according to rules notification must not be send.

KVGrygoriev, 2018-05-14:
looks like it has already fixed

[16:25:22,573] OnButtonPress_InBackgoundHmiLevel_IsIgnored_name_OK_mode_SHORT [SUCCESS] (2107 ms)
[16:25:24,750] OnButtonPress_InBackgoundHmiLevel_IsIgnored_name_OK_mode_LONG [SUCCESS] (2062 ms)
[16:25:26,871] OnButtonPress_InBackgoundHmiLevel_IsIgnored_name_SEEKLEFT_mode_SHORT [SUCCESS] (1925 ms)
[16:25:28,872] OnButtonPress_InBackgoundHmiLevel_IsIgnored_name_SEEKLEFT_mode_LONG [SUCCESS] (1936 ms)
[16:25:30,871] OnButtonPress_InBackgoundHmiLevel_IsIgnored_name_SEEKRIGHT_mode_SHORT [SUCCESS] (1930 ms)
[16:25:32,874] OnButtonPress_InBackgoundHmiLevel_IsIgnored_name_SEEKRIGHT_mode_LONG [SUCCESS] (1913 ms)
[16:25:34,870] OnButtonPress_InBackgoundHmiLevel_IsIgnored_name_TUNEUP_mode_SHORT [SUCCESS] (1933 ms)
[16:25:36,870] OnButtonPress_InBackgoundHmiLevel_IsIgnored_name_TUNEUP_mode_LONG [SUCCESS] (1930 ms)
[16:25:38,869] OnButtonPress_InBackgoundHmiLevel_IsIgnored_name_TUNEDOWN_mode_SHORT [SUCCESS] (1930 ms)
[16:25:40,869] OnButtonPress_InBackgoundHmiLevel_IsIgnored_name_TUNEDOWN_mode_LONG [SUCCESS] (1932 ms)
[16:25:42,870] OnButtonPress_InBackgoundHmiLevel_IsIgnored_name_PRESET_0_mode_SHORT [SUCCESS] (1929 ms)
[16:25:44,869] OnButtonPress_InBackgoundHmiLevel_IsIgnored_name_PRESET_0_mode_LONG [SUCCESS] (1929 ms)
[16:25:46,869] OnButtonPress_InBackgoundHmiLevel_IsIgnored_name_PRESET_1_mode_SHORT [SUCCESS] (1929 ms)
[16:25:48,871] OnButtonPress_InBackgoundHmiLevel_IsIgnored_name_PRESET_1_mode_LONG [SUCCESS] (1920 ms)
[16:25:50,868] OnButtonPress_InBackgoundHmiLevel_IsIgnored_name_PRESET_2_mode_SHORT [SUCCESS] (1931 ms)
[16:25:52,869] OnButtonPress_InBackgoundHmiLevel_IsIgnored_name_PRESET_2_mode_LONG [SUCCESS] (1906 ms)
[16:25:54,871] OnButtonPress_InBackgoundHmiLevel_IsIgnored_name_PRESET_3_mode_SHORT [SUCCESS] (1928 ms)
[16:25:56,869] OnButtonPress_InBackgoundHmiLevel_IsIgnored_name_PRESET_3_mode_LONG [SUCCESS] (1926 ms)
[16:25:58,870] OnButtonPress_InBackgoundHmiLevel_IsIgnored_name_PRESET_4_mode_SHORT [SUCCESS] (1926 ms)
[16:26:00,869] OnButtonPress_InBackgoundHmiLevel_IsIgnored_name_PRESET_4_mode_LONG [SUCCESS] (1931 ms)
[16:26:02,869] OnButtonPress_InBackgoundHmiLevel_IsIgnored_name_PRESET_5_mode_SHORT [SUCCESS] (1926 ms)
[16:26:04,870] OnButtonPress_InBackgoundHmiLevel_IsIgnored_name_PRESET_5_mode_LONG [SUCCESS] (1915 ms)
[16:26:06,872] OnButtonPress_InBackgoundHmiLevel_IsIgnored_name_PRESET_6_mode_SHORT [SUCCESS] (1915 ms)
[16:26:08,872] OnButtonPress_InBackgoundHmiLevel_IsIgnored_name_PRESET_6_mode_LONG [SUCCESS] (1910 ms)
[16:26:10,871] OnButtonPress_InBackgoundHmiLevel_IsIgnored_name_PRESET_7_mode_SHORT [SUCCESS] (1923 ms)
[16:26:12,869] OnButtonPress_InBackgoundHmiLevel_IsIgnored_name_PRESET_7_mode_LONG [SUCCESS] (1922 ms)
[16:26:14,869] OnButtonPress_InBackgoundHmiLevel_IsIgnored_name_PRESET_8_mode_SHORT [SUCCESS] (1922 ms)
[16:26:16,870] OnButtonPress_InBackgoundHmiLevel_IsIgnored_name_PRESET_8_mode_LONG [SUCCESS] (1933 ms)
[16:26:18,872] OnButtonPress_InBackgoundHmiLevel_IsIgnored_name_PRESET_9_mode_SHORT [SUCCESS] (1919 ms)
[16:26:20,869] OnButtonPress_InBackgoundHmiLevel_IsIgnored_name_PRESET_9_mode_LONG [SUCCESS] (1934 ms)
[16:26:22,874] OnButtonPress_InBackgoundHmiLevel_IsIgnored_name_SEARCH_mode_SHORT [SUCCESS] (1908 ms)
[16:26:24,876] OnButtonPress_InBackgoundHmiLevel_IsIgnored_name_SEARCH_mode_LONG [SUCCESS] (1907 ms)
[16:26:26,879] OnButtonPress_InBackgoundHmiLevel_IsIgnored_name_CUSTOM_BUTTON_mode_SHORT_customBu... [SUCCESS] (1908 ms)
[16:26:28,875] OnButtonPress_InBackgoundHmiLevel_IsIgnored_name_CUSTOM_BUTTON_mode_SHORT_customBu... [SUCCESS] (1908 ms)
[16:26:30,869] OnButtonPress_InBackgoundHmiLevel_IsIgnored_name_CUSTOM_BUTTON_mode_LONG_customBut... [SUCCESS] (1927 ms)
[16:26:32,869] OnButtonPress_InBackgoundHmiLevel_IsIgnored_name_CUSTOM_BUTTON_mode_LONG_customBut... [SUCCESS] (1928 ms)

closed 2018-05-14 by KVGrygoriev