smartdevicelink/sdl_core #964: SDL sends to mobile "APPLICATION_NOT_REGISTRED" in setAppIcon responce if HMI responds with "INVALID_DATA"

KhrystynaDubovyk, 2016-11-07:
**Occurrence:**
Always

**Precondition:**
SDL Core and HMI are started. App is registered, HMI level = FULL

**Steps:**
1. Send PutFile with file name =\syncFileName
SDL returns SUCCESS result, success:true
2. Send SetAppIcon with file name =\syncFileName

**Actual result:**
-HMI tries to find image from the path: file:///root/develop22/bin/storage/123QWE_9f2973caa2a417e9e0cacd6d29dcaa514b977130031396b997c85b94304900c3//syncFileName
and couldn't find the any file then returns code 11 ("INVALID_DATA") to SDL (known HMI defect APPLINK-22908)
-SDL returns "APPLICATION_NOT_REGISTRED" to mobile 

**Expected result:**
-HMI tries to find image from the path: file:///root/develop22/bin/storage/123QWE_9f2973caa2a417e9e0cacd6d29dcaa514b977130031396b997c85b94304900c3/\syncFileName
then finds the file and then returns code 0 to SDL.
-SDL should send to mobile result code which was sent to SDL by HMI

**Enviroment:**
SDL branch: a6c32b45832653d6b851be9b19355fa35f377e07
HMI: HTML5
ATF version: -
ATF repo: -
ATF branch: -
Scripts repo: -
Scripts branch: -
OS: Ubuntu 14.04.01
Transport: Wi-Fi
Mobile device: Lenovo P1ma40
Mobile OS: Android V5.1
Mobile App version: Luxoft SPT
Mobile App type (media/non-media/navi/any): any
Virtual Machine using (yes/no): no

**Attachments:**

vvvakulenko, 2016-12-20:
PASSED
was verified on:
SDL repo: https://github.com/smartdevicelink/sdl_core/
SDL branch: a6c32b45832653d6b851be9b19355fa35f377e07
HMI: HTML5
ATF version: -
ATF repo: - 
ATF branch: - 
Scripts repo: - 
Scripts branch: -
OS: Ubuntu 14.04
Transport: Wi-Fi
Mobile device: Huawei U9200
Mobile OS: Android V4.0.3
Mobile App version: Luxoft SPT
Mobile App type (media/non-media/navi/any): any
Virtual Machine using (yes/no): yes

jacobkeeler, 2017-04-18:
@vvvakulenko Does your comment mean that this issue is already fixed in 4.2?

LitvinenkoIra, 2017-12-01:
Priority Medium.
Unregistration app in standard use case.

HSavynetska, 2018-05-17:
The defect is not reproduced
sdl_core: branch develop
commit: 65c1e7eb77fa949cce9c07f229e2cf747f297e14
Verified with script:
Script is under review
https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/1897

HSavynetska, 2018-05-21:
The defect is not reproduced
sdl_core: branch develop
commit: 65c1e7e
Verified with script:
Script is under review
https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/1903

AStasiuk, 2018-06-07:
Closed as not reproducible.

closed 2018-06-07 by AStasiuk

GetmanetsIrina, 2018-07-30:
Script for reproducing https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Defects/5_0/964_SDL_sends_to_mobile_APPLICATION_NOT_REGISTRED_in_setAppIcon_responce_if_HMI_responds_with_INVALID_DATA.lua