smartdevicelink/sdl_core #965: [API] App is not unregistered by reason = REQUEST_WHILE_IN_NONE_HMI_LEVEL

DrachenkoAnastasiia, 2016-11-07:
**Occurrence:**
Always

**Preconditions:**
- SDL Core and HMI are started.
- App is registered
- Set AppHMILevelNoneTimeScaleMaxRequests= 1 in the ini file

**Steps to reproduce:**
1. Send 10 ListFiles request from app
**Note:** Should use the attached lua file to reproduce this defect automatically

**Actual result:**
App is not unregistered by REQUEST_WHILE_IN_NONE_HMI_LEVEL reason

**Expected result:**
App is unregistered by REQUEST_WHILE_IN_NONE_HMI_LEVEL reason

**Environment:**
Reproduces
Release: 4.2.0 (master marge)
Repository: https://github.com/smartdevicelink/sdl_core
Commit: a6c32b45832653d6b851be9b19355fa35f377e07
OS: Ubuntu 14.04

**Attachments:**

LitvinenkoIra, 2017-12-01:
Priority Medium.
Loss of functionality on SDL side with standard use-case.

JackLivio, 2018-03-06:
Verified exists on develop

HSavynetska, 2018-05-21:
The defect is not reproduced
sdl_core: branch develop
commit: 65c1e7e
Verified with script:
Script is under review
#1902

AStasiuk, 2018-06-07:
Closed as not reproducible.

closed 2018-06-07 by AStasiuk

GetmanetsIrina, 2018-07-30:
Script for reproducing https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Defects/5_0/965_API_App_is_not_unregistered_by_reason_%3D_REQUEST_WHILE_IN_NONE_HMI_LEVELlua